jiangcuo/pxvirt #1: New repository malformed?

eyduh, 2023-07-24:
I'm having some trouble adding the new repository.

Since the key and release files are at different paths apt has some trouble accepting it:

```
mkdir /etc/apt/keyrings

wget -O- https://global.mirrors.apqa.cn/proxmox/debian/pveport.gpg | gpg --dearmor | tee /etc/apt/keyrings/pveport.gpg

echo "deb [signed-by=/etc/apt/keyrings/pveport.gpg] https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye port" | tee /etc/apt/sources.d/pveport.list

apt update

...
Ign:6 https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye InRelease
Get:7 https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye Release [907 B]
Ign:8 https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye Release.gpg
Reading package lists... Done
E: The repository 'https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

If I instead point the repo to 
`https://global.mirrors.apqa.cn/proxmox/debian` which is where the gpg is located it instead complains about not finding the appropriate Release file.

Am I missing some step in setting it up?

lextor007, 2023-07-25:
Hi, 

I’m having the same issue after updated to the new repo.

Repo is not signed.

jiangcuo, 2023-07-25:
Hi all , it cause by rsync delay.
I'm distributing repositories in multiple regions, and some auto-sync scripts haven't been created yet.
Now global is ok

lextor007, 2023-07-25:
Hi jiangcuo,

tested again and it’s working for me.
Thanks!

eyduh, 2023-07-26:
Awesome! works now c:

closed 2023-07-26 by eyduh