jiangcuo/pxvirt #2: Unable to install proxmox-ve

kenji21, 2023-08-27:
```
Selecting previously unselected package proxmox-ve.
(Reading database ... 52583 files and directories currently installed.)
Preparing to unpack .../proxmox-ve_8.0.1_all.deb ...
Unpacking proxmox-ve (8.0.1) ...
Setting up proxmox-ve (8.0.1) ...
ln: failed to create symbolic link '/boot/pve/vmlinuz': Operation not permitted
ln: failed to create symbolic link '/boot/pve/initrd.img': Operation not permitted
```

when `apt install proxmox-ve` on a Pi4 compute module

jiangcuo, 2023-08-28:
Ignore it.

kenji21, 2023-08-28:
ok, but the kernel is still the "raspbian" one, doesn't it lack features needed by proxmox ?

jiangcuo, 2023-08-28:
You can use the raspbian kernel If you don't use zfs,

kenji21, 2023-08-28:
I was planing to use zfs

jiangcuo, 2023-08-28:
You can build kernel by yourself ,visit  https://github.com/jiangcuo/pve-port-kernel.git

Or use prebuild kernel. https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md
add kernel repo and ```apt update && apt search pve-kernl```, 6.1 kernel is LTS and recommend.

kenji21, 2023-08-31:
Ended up installing zfs using dkms, but had to :
```
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230405-1_arm64.deb
dpkg -i --force-all raspberrypi-kernel-headers_1.20230405-1_arm64.deb
```
(from https://raspberrypi.stackexchange.com/questions/143009/kernel-and-header-version-not-the-same)

before `apt install zfs-dkms zfsutils-linux` (from https://www.jeffgeerling.com/blog/2021/htgwa-create-zfs-raidz1-zpool-on-raspberry-pi)

closed 2023-08-31 by kenji21

kenji21, 2023-08-31:
Ok, unable to create a container....
```
root@pi4:~# pct create 1002 /var/lib/vz/template/cache/debian-12-standard_arm64.tar.xz -description testsarmpi -rootfs 16 -hostname testsarmpi -features nesting=1 -unprivileged 1 -net0 name=eth0,hwaddr=02:13:90:35:ec:8c,ip=192.168.63.12/24,gw=192.168.63.254,bridge=vmbr0 -storage zfs-storage -memory 4096 -onboot yes
extracting archive '/var/lib/vz/template/cache/debian-12-standard_arm64.tar.xz'
Total bytes read: 451840000 (431MiB, 25MiB/s)
Detected container architecture: arm64
unable to open file '/etc/network/interfaces.tmp.16260' - No such file or directory
unable to create CT 1002 - error in setup task PVE::LXC::Setup::post_create_hook
```