jiangcuo/pxvirt #3: unable to create a container....

jiangcuo, 2023-09-02:
Ok, unable to create a container....
```
root@pi4:~# pct create 1002 /var/lib/vz/template/cache/debian-12-standard_arm64.tar.xz -description testsarmpi -rootfs 16 -hostname testsarmpi -features nesting=1 -unprivileged 1 -net0 name=eth0,hwaddr=02:13:90:35:ec:8c,ip=192.168.63.12/24,gw=192.168.63.254,bridge=vmbr0 -storage zfs-storage -memory 4096 -onboot yes
extracting archive '/var/lib/vz/template/cache/debian-12-standard_arm64.tar.xz'
Total bytes read: 451840000 (431MiB, 25MiB/s)
Detected container architecture: arm64
unable to open file '/etc/network/interfaces.tmp.16260' - No such file or directory
unable to create CT 1002 - error in setup task PVE::LXC::Setup::post_create_hook
```

_Originally posted by @kenji21 in https://github.com/jiangcuo/Proxmox-Port/issues/2#issuecomment-1701672554_

jiangcuo, 2023-09-02:
debian12 using systemd-network .

kenji21, 2023-09-03:
same issue with debian 11 container :
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    436      0 --:--:-- --:--:-- --:--:--   438
100 71.5M  100 71.5M    0     0  15.5M      0  0:00:04  0:00:04 --:--:-- 18.4M

root@pi4:~# pct  create 1002 /var/lib/vz/template/cache/debian-11-standard_arm64.tar.xz -description testsarmpi -rootfs 16 -hostname testsarmpi -features nesting=1 -unprivileged 1 -net0 name=eth0,hwaddr=02:13:90:35:ec:8c,ip=192.168.63.12/24,gw=192.168.63.254,bridge=vmbr0 -storage zfs-storage -memory 4096 -onboot yes
extracting archive '/var/lib/vz/template/cache/debian-11-standard_arm64.tar.xz'
Total bytes read: 344678400 (329MiB, 23MiB/s)
Detected container architecture: arm64
unable to open file '/etc/network/interfaces.tmp.104204' - No such file or directory
unable to create CT 1002 - error in setup task PVE::LXC::Setup::post_create_hook
```

kenji21, 2023-09-03:
I think it's the interface creation on host that is failing

jiangcuo, 2023-09-03:
follow wiki and download lxc form apqa.
https://github.com/jiangcuo/Proxmox-Arm64/wiki/Linux-Container

I have tested.

kenji21, 2023-09-04:
without `-net0 name=eth0,ip=....` option, container (with rootfs.tar.xz from images.linuxcontainers.org), container is created

kenji21, 2023-09-04:
as stated in your wiki and in this thread:
https://discuss.linuxcontainers.org/t/pimox7-rpi4-arm64-ct-basic-configuration/13996/9

edit `/etc/systemd/network/eth0.network` to set ip/gateway:
```
[Network]
Address=192.168.1.101/24
Gateway=192.168.1.1
```
and restart ` systemctl restart systemd-networkd`

kenji21, 2023-09-04:
ok, fine if container is already booted, but not if adding network before booting it:
```
lxc-start -F -n 120  --logfile=lxc.log --logpriority=debug
```

```
lxc-start 1002 20230904042918.435 INFO     conf - ../src/lxc/conf.c:run_script_argv:338 - Executing script "/usr/share/lxc/hooks/lxc-pve-prestart-hook" for container "1002", config section "lxc"
lxc-start 1002 20230904042920.361 DEBUG    conf - ../src/lxc/conf.c:run_buffer:311 - Script exec /usr/share/lxc/hooks/lxc-pve-prestart-hook 1002 lxc pre-start produced output: unable to open file '/etc/network/interfaces.tmp.267919' - No such file or directory

lxc-start 1002 20230904042920.369 DEBUG    conf - ../src/lxc/conf.c:run_buffer:311 - Script exec /usr/share/lxc/hooks/lxc-pve-prestart-hook 1002 lxc pre-start produced output: error in setup task PVE::LXC::Setup::pre_start_hook

lxc-start 1002 20230904042920.416 ERROR    conf - ../src/lxc/conf.c:run_buffer:322 - Script exited with status 1
lxc-start 1002 20230904042920.416 ERROR    start - ../src/lxc/start.c:lxc_init:844 - Failed to run lxc.hook.pre-start for container "1002"
```

kenji21, 2023-09-05:
definitely an image issue, using devuan (debian fork without systemd) is OK:

```
root@pi4:/home/kenji# curl -L https://uk.lxd.images.canonical.com/images/devuan/daedalus/arm64/default/20230830_11:50/rootfs.tar.xz -o /var/lib/vz/template/cache/devuan-12-standard_arm64.tar.xz 2>/dev/null

root@pi4:/home/kenji# pct create 1003 /var/lib/vz/template/cache/devuan-12-standard_arm64.tar.xz -description devuan -rootfs 16 -hostname devuan -features nesting=1 -unprivileged 1 -net0 name=eth0,hwaddr=02:13:90:35:ec:8d,ip=192.168.63.13/24,gw=192.168.63.254,bridge=vmbr0 -storage zfs-storage -memory 4096 -onboot yes
extracting archive '/var/lib/vz/template/cache/devuan-12-standard_arm64.tar.xz'
Total bytes read: 394526720 (377MiB, 24MiB/s)
Detected container architecture: arm64
Creating SSH host key 'ssh_host_ecdsa_key' - this may take some time ...
done: SHA256:wCKHYlBIxEXDXdo/2idRZgHDJRmwPPnoUKVbZ0nyuLA root@devuan
Creating SSH host key 'ssh_host_rsa_key' - this may take some time ...
done: SHA256:gxru3GOv2gqKjTVXCQBFcbgczHLE3MNElDqtuwKJOW0 root@devuan
Creating SSH host key 'ssh_host_dsa_key' - this may take some time ...
done: SHA256:mWEzikYbJGvhuyfdYyDyw6oS9kZdgZvxLEeB8BfhJHo root@devuan
Creating SSH host key 'ssh_host_ed25519_key' - this may take some time ...
done: SHA256:1p7egjyVJ8whOILGE5QlHgzZCyOjQ7YoK4MNVZyKH+I root@devuan

root@pi4:/home/kenji# pct start 1003
root@pi4:/home/kenji# pct enter 1003

root@devuan:/# ping google.fr
PING google.fr(par10s41-in-x03.1e100.net (2a00:1450:4007:80d::2003)) 56 data bytes
64 bytes from par10s41-in-x03.1e100.net (2a00:1450:4007:80d::2003): icmp_seq=1 ttl=113 time=10.5 ms
^C
--- google.fr ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 10.531/10.531/10.531/0.000 ms

root@devuan:/# ping4 google.fr
PING  (142.250.75.227) 56(84) bytes of data.
64 bytes from par10s41-in-f3.1e100.net (142.250.75.227): icmp_seq=1 ttl=118 time=10.3 ms
64 bytes from par10s41-in-f3.1e100.net (142.250.75.227): icmp_seq=2 ttl=118 time=9.91 ms
^C
---  ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
```

kenji21, 2023-09-05:
> debian12 using systemd-network .

it is disabled within proxmox container setup: 

```
root@pi4:~# grep -A 7 -B 12 systemd-networkd /usr/share/perl5/PVE/LXC/Setup/Debian.pm

sub setup_init {
    my ($self, $conf) = @_;

    my $systemd = $self->ct_readlink('/sbin/init');
    if (defined($systemd) && $systemd =~ m@/systemd$@) {
	$self->setup_container_getty_service($conf);

	my $version = $self->{version};
	if (at_least($version, 12, 0)) {
	    # this only affects the first-boot (if no /etc/machien-id exists).
	    $self->setup_systemd_preset({
		# systemd-networkd gets enabled by default, disable it, debian uses
		# ifupdown
		'systemd-networkd.service' => 0,
	    });
	}
    }

    setup_inittab($self, $conf);
}
```

kenji21, 2023-09-05:
there is a `setup_systemd_networkd` function in `/usr/share/perl5/PVE/LXC/Setup/Base.pm`:

```
root@pi4:~# grep -R setup_systemd_networkd  /usr/share/perl5/
/usr/share/perl5/PVE/LXC/Setup/ArchLinux.pm:    $self->setup_systemd_networkd($conf);
/usr/share/perl5/PVE/LXC/Setup/Ubuntu.pm:	$self->setup_systemd_networkd($conf);
/usr/share/perl5/PVE/LXC/Setup/NixOS.pm:    $self->setup_systemd_networkd($conf);
/usr/share/perl5/PVE/LXC/Setup/Fedora.pm:    $self->SUPER::setup_systemd_networkd($conf) if $setup_systemd;
/usr/share/perl5/PVE/LXC/Setup/Base.pm:sub setup_systemd_networkd {
```

```
sub setup_systemd_networkd {
    my ($self, $conf) = @_;

    foreach my $k (keys %$conf) {
        next if $k !~ m/^net(\d+)$/;
        my $d = PVE::LXC::Config->parse_lxc_network($conf->{$k});
        next if !$d->{name};

        my $filename = "/etc/systemd/network/$d->{name}.network";

        my $data = <<"DATA";
[Match]
Name = $d->{name}

[Network]
Description = Interface $d->{name} autoconfigured by PVE
DATA

        my $routes = '';
        my ($has_ipv4, $has_ipv6);

```

kenji21, 2023-09-05:
Ok, I fixed container creation with by adding this function to /usr/share/perl5/PVE/LXC/Setup/Debian.pm:

```

sub template_fixup {
    my ($self, $conf) = @_;

    $self->ct_make_path('/etc/network');
}

```

But as the container image is lacking ifupdown package, had to `pct enter 1004` to do:

```
root@testsarmpi4:~# ifconfig eth0 192.168.63.14/24 192.168.63.254 up
root@testsarmpi4:~# ip route
default via 192.168.63.254 dev eth0 proto dhcp src 192.168.63.131 metric 1024
192.168.63.0/24 dev eth0 proto kernel scope link src 192.168.63.131 metric 1024
192.168.63.254 dev eth0 proto dhcp scope link src 192.168.63.131 metric 1024

root@testsarmpi4:~# apt install ifupdown
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  ppp rdnssd
The following NEW packages will be installed:
  ifupdown
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 60.1 kB of archives.
After this operation, 252 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 ifupdown arm64 0.8.41 [60.1 kB]
Fetched 60.1 kB in 0s (685 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package ifupdown.
(Reading database ... 13574 files and directories currently installed.)
Preparing to unpack .../ifupdown_0.8.41_arm64.deb ...
Unpacking ifupdown (0.8.41) ...
Setting up ifupdown (0.8.41) ...
Created symlink /etc/systemd/system/multi-user.target.wants/networking.service → /lib/systemd/system/networking.service.
Created symlink /etc/systemd/system/network-online.target.wants/networking.service → /lib/systemd/system/networking.service.

root@testsarmpi4:~# reboot
```

Network is operational after reboot

kenji21, 2023-09-05:
related to : https://github.com/lxc/distrobuilder/issues/615#issuecomment-1109099473

jiangcuo, 2023-09-05:
Thx. I make a mistake. 
"When configuring the CT, leave IP = Static. Do not enter any IP address."
I always use ip=none when creating the ct.This gave me a false impression that  there was no problem.

This should be the upstream issue.Debian bullseye, bookworm and sid now use systemd-networkd.

closed 2023-12-12 by jiangcuo

Anexgohan, 2023-12-23:
### Here is a solution after browsing here and other forums:

Debian lxc creation throws error, where location/files in /etc/network/interfaces are missing, debian 12 and 11 uses systemd "/etc/systemd/network/eth0.network"
Ubuntu mantic lxc works without issues.

Fix:
1). Create the debian lxc/ct, select static on the network tab and leave it blank.
2). Start the lxc and login, should be unable to get an IP address
`ip -c a`
3). run the following command to get an IP address
`dhclient`
4). run to update repos
`apt update`
5). to install ifupdown2 package, Note: if this doesn't work out try "apt install ifupdown"
`apt install ifupdown2 -y`
6). upgrade everything
`apt upgrade -y`

7). Shut down the lxc and goto its network tab, in the ip4 section select DHCP and confirm ok
8). Start the lxc and you should get an IP without issue automatically

Thanks to all the above people and proxmox forum for this solution.

or

### **a single copy/paste command for the lazy**

```
dhclient && \
apt update && \
apt install ifupdown2 -y && \
apt upgrade -y

martink1979, 2023-12-26:
@Anexgohan  Super, damit hat es funktioniert ! Aber was ist denn das Problem dahinter?  Ich habe eine Debian 10 "Buster" 
ohne Probleme als CT Container installieren können, gleich mit IP Vergabe.. aber wenn ich die gleiche Installationsroutine mit einem Debian Bullseye oder Bookworm mache, kommt der Fehler ?

MeikelLP, 2024-06-26:
For anyone stumbling across this too:

I had a fresh proxmox installation but i was missing the package `ifupdown2` which caused this behavior. Just install it:

```
apt install ifupdown2 -y
```

kenji21, 2024-07-05:
The root fix should be Proxmox to manage ifupdown/systemd managed network with file checks

biplobsd, 2024-08-28:
> ### Here is a solution after browsing here and other forums:
> Debian lxc creation throws error, where location/files in /etc/network/interfaces are missing, debian 12 and 11 uses systemd "/etc/systemd/network/eth0.network" Ubuntu mantic lxc works without issues.
> 
> ### Fix: (_Run this in lxc shell_)
> 1). Create the debian lxc/ct, select static on the network tab and leave it blank. 2). Start the lxc and login, should be unable to get an IP address `ip -c a` 3). run the following command to get an IP address `dhclient` 4). run to update repos `apt update` 5). to install ifupdown2 package, Note: if this doesn't work out try "apt install ifupdown" `apt install ifupdown2 -y` 6). upgrade everything `apt upgrade -y`
> 
> 7). Shut down the lxc and goto its network tab, in the ip4 section select DHCP and confirm ok 8). Start the lxc and you should get an IP without issue automatically
> 
> Thanks to all the above people and proxmox forum for this solution.
> 
> or
> 
> ### **Single copy/paste command for the lazy**
> ```
> dhclient && \
> apt update && \
> apt install ifupdown2 -y && \
> apt upgrade -y && \
> ip -c a
> ```
> 
> Shut down lxc, goto its network tab, , in the ip4 section select DHCP Start the LXC
> 
> NOTE: If you have two IP addresses, you have to disable the following: `nano /etc/systemd/network/eth0.network` put a '#' in front of every line to disable this file completely Restart the lxc

Thanks & worked