jiangcuo/pxvirt #5: zfs not loading

kinhsman, 2023-11-01:
Hi, I have Proxmox 8.0.6 running on NanoPi R6S. The issue i'm running into is that I cannot create a ZFS pool. Any insight is greatly appreciated!
[image]

```
root@NanoPi-PVE:~# zpool list
The ZFS modules are not loaded.
Try running '/sbin/modprobe zfs' as root to load them.
root@NanoPi-PVE:~# /sbin/modprobe zfs
modprobe: FATAL: Module zfs not found in directory /lib/modules/5.10.160
root@NanoPi-PVE:~# 
```

jiangcuo, 2023-11-04:
https://github.com/jiangcuo/Proxmox-Arm64/issues/45

closed 2023-11-21 by jiangcuo

LuisPuertollano, 2024-12-19:
Hello Jiangcuo, hello evryone,

I`ve followed all linked ressources as resolution but I'm still stucked ...is there any way to have a unified explanation?

actual status is following, I'm triying to create a ZFS Pool for 4x4TB SSDs
[image]