jiangcuo/pxvirt #6: Can't restore on pve 8.0.6 on raspbery pi 5 from Proxmox Backup Server

Thimoty44, 2023-11-12:
Hi, everyone, the pve on arm product is awesome so I have create a vm to test proxmox backup server with arm too .
I backed up a test vm on the promox backup server arm and tried to restore it but I have meet the following error :

 Logical volume "vm-100-disk-1" successfully removed.
  Logical volume "vm-100-disk-0" successfully removed.
  Logical volume "vm-100-disk-0" created.
new volume ID is 'pimox-storage:vm-100-disk-0'
  Logical volume "vm-100-disk-1" created.
new volume ID is 'pimox-storage:vm-100-disk-1'
restore proxmox backup image: /usr/bin/pbs-restore --repository backup@pbs@192.168.50.19:TestPBS vm/100/2023-11-12T12:00:35Z drive-efidisk0.img.fidx /dev/vg-pimox/vm-100-disk-0 --verbose --format raw --skip-zero
connecting to repository 'backup@pbs@192.168.50.19:TestPBS'
open block backend for target '/dev/vg-pimox/vm-100-disk-0'
starting to restore snapshot 'vm/100/2023-11-12T12:00:35Z'
download and verify backup index
restore failed: unable to read fixed index 'drive-efidisk0.img.fidx' - EINVAL: Invalid argument
  Logical volume "vm-100-disk-0" successfully removed.
temporary volume 'pimox-storage:vm-100-disk-0' sucessfuly removed
  Logical volume "vm-100-disk-1" successfully removed.
temporary volume 'pimox-storage:vm-100-disk-1' sucessfuly removed
error before or during data restore, some or all disks were not completely restored. VM 100 state is NOT cleaned up.
TASK ERROR: command '/usr/bin/pbs-restore --repository backup@pbs@192.168.50.19:TestPBS vm/100/2023-11-12T12:00:35Z drive-efidisk0.img.fidx /dev/vg-pimox/vm-100-disk-0 --verbose --format raw --skip-zero' failed: exit code 255

This is possibly a bug because I can backup but not restore on pve 8.0.6 on arm on raspbery pi 5.
Could someone help me with that ?

Have a nice day.

Thimoty44, 2023-11-13:
I just figured out when I go in file restore on pve I got this error :
cannot run file-restore VM: package 'proxmox-backup-restore-image' is not (correctly) installed (500)

So I tried reinstalled with this command :
apt reinstall proxmox-backup-client proxmox-backup-file-restore --no-install-recommends

But it did not work

jiangcuo, 2023-11-15:
1. What's your pbs hardware and version.
2. `proxmox-backup-restore-image` is built for x86_64 and I'm trying to port it. I'll reply if there's progress.

Thimoty44, 2023-11-15:
Hi my pbs version is 3.0-2 from repository : deb https://global.mirrors.apqa.cn/proxmox/debian/pbs bookworm port
I am using debian 12 for arm 
This is a VM create from my pve on arm 
This my conf for the PBS vm :
[image]

Thimoty44, 2023-11-21:
Hi, after reinstalling for file restore I got the following error :
[image]

With restore same error : 
[image]

I tried this test with a new vm debian 12 arm that I backed up in stop mode after reinstalling packages then tried to restore without sucess.
Have a nice day, I hope this errors could help you.

jiangcuo, 2023-11-21:
Do you have an x86 machine？
You can use an x86 PBS to backup the ARM VM and restore .
Or backup the x86 VM and restore for test.
Or  install pbs on your host and backup or restore VM.

Thimoty44, 2023-11-23:
> Do you have an x86 machine？ You can use an x86 PBS to backup the ARM VM and restore . Or backup the x86 VM and restore for test. Or install pbs on your host and backup or restore VM.

Hi, sorry I don't have any vm in x86 or x86 machine but I tried something this a workaround currently if you have a PBS you can mount an NFS on your PBS or any other os and mount it in proxmox : 
[image]

And you can restore with the pve : 
[image]

The only thing is that I had to mount nfs manually in /etc/fstab : 
[image]

I added storage by using the directory mode in proxmox : 
[image]

If u have a problem mounting nfs when rebooting do this in your crontab : 
@reboot /usr/bin/sleep 60 && /usr/bin/systemctl daemon-reload && /usr/bin/mount -a

The only things with this work around we loose the prune and compression of PBS we only use the compression used by the proxmox.
For pruning configure over you backup on Proxmox and when you add the storage as directory.

Hope this workaround could help fort other have a nice day guys.

jiangcuo, 2023-11-23:
I can't reproduce your problem, there is no problem with the pbs of arm64 on my side.
`file-restore`  has fixed  and It requires the kernel to support vsock.

closed 2023-12-12 by jiangcuo

jiangcuo, 2023-12-24:
PBS need 4k page size.