jiangcuo/pxvirt #7: Risc-V Support

liberodark, 2023-11-16:
Hi,

Im on Risc-V and ask you for try to make a port of PVE & PBS on `riscv64` is going to be fully supported on Debian 13.
If you've done well with me for tests I'm available.

Best Regards

jiangcuo, 2023-11-16:
Riscv64 only works with 6.4.If you're interested, I can upload a package from bookworm.

liberodark, 2023-11-16:
Hi,

Yes if you have possibility to upload bookworm that would be great.
Actually im on Trixie (Sid) for the official support of Debian.

Best Regards

liberodark, 2023-11-16:
Have try to set : 

But you don't have `riscv64` arch supported on this repo.
Also can you push `riscv64` in sid directly ?

```
deb [arch=riscv64] https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm port
deb [arch=riscv64] https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm ceph-reef
deb [arch=riscv64] https://global.mirrors.apqa.cn/proxmox/debian/pbs bookworm port
```

jiangcuo, 2023-11-17:
Hi,
I've uploaded the package to the sid directory.
`deb https://global.mirrors.apqa.cn/proxmox/debian/pve sid port`.
It works on my starfive2 (accel=tcg).
For vm configurations, use `cpu: rv64` and  `bios: ovmf`

liberodark, 2023-11-17:
Hi,

Thanks for the upload in sid.
Im lock at :   `ln: failed to create symbolic link '/boot/pve/vmlinuz': Operation not permitted`

jiangcuo, 2023-11-17:
FAT does not support soft link.
RISCV is a beta version and many features are not available.