jiangcuo/pxvirt #10: PBS and PVE Repositories Unavailable for Over 24 Hours

D4rk4, 2023-11-24:
For the past 24+ hours, we have been experiencing issues accessing the PBS and PVE repositories. Attempts to fetch data from these repositories are consistently failing, impacting our ability to install or update Proxmox. Below are the error messages we're encountering:

```
E: Failed to fetch http://global.mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/InRelease  521   [IP: 172.67.186.225 80]

E: Failed to fetch http://global.mirrors.apqa.cn/proxmox/debian/pbs/dists/bookworm/InRelease  521   [IP: 172.67.186.225 80]
```

- [ ] Issue Resolution: Please investigate and resolve the issue preventing access to the PBS and PVE repositories.
- [ ] Repository Mirroring: Additionally, we request guidance or a method to mirror these repositories. This would serve as a contingency plan to maintain access to necessary resources in the event of future outages of the origin server.

Your prompt attention to this matter would be greatly appreciated, as it is critical for our ongoing operations and maintenance of Proxmox systems.

jiangcuo, 2023-11-24:
Hello fix it.
Switched to Cloudflare yesterday and blocked HTTP requests. Http and https both work now.
You can visit `https://mirrors.apqa.cn/` to see the mirror list

D4rk4, 2023-11-24:
@jiangcuo please update the documentation

closed 2023-11-24 by D4rk4