jiangcuo/pxvirt #14: sound support in vm

simper, 2023-12-17:
I see spice is not supported in arm proxmox, but to add audio into VM, I can see the only backend driver option available is spice, does that mean there is no way to have sound support in vm for arm64 proxmox?

jiangcuo, 2023-12-18:
Qemu not support qxl now.
There is two ways to easily connect  to a vm using spice .
1. Set "vga: virtio-gl" to active spice.
2. Add args "-spice port=22222,disable-ticketing" to connect by manual.

simper, 2023-12-21:
Tried to configure virtio-gl in VM config, then start VM, I got below error:

> TASK ERROR: missing libraries for 'virtio-gl' detected! Please install 'libgl1' and 'libegl1'

But actually those libs were installed on proxmox host:

> libgl1/stable,now 1.6.0-1 arm64 [**installed**]
>   Vendor neutral GL dispatch library -- legacy GL support

> libegl1/stable,now 1.6.0-1 arm64 [installed]
>   Vendor neutral GL dispatch library -- EGL support

No idea what does the error message mean?

jiangcuo, 2023-12-21:
Hi,
There has a fix patch.
https://github.com/jiangcuo/qemu-server/commit/acee120e778d66787007a43cedc7203aa790a320
You can edit ` /usr/share/perl5/PVE/QemuServer.pm` by manual

simper, 2023-12-21:
> Hi, There has a fix patch. [jiangcuo/qemu-server@acee120](https://github.com/jiangcuo/qemu-server/commit/acee120e778d66787007a43cedc7203aa790a320) You can edit ` /usr/share/perl5/PVE/QemuServer.pm` by manual

Tried the quick fix, but new error pop up:

> error: XDG_RUNTIME_DIR is invalid or not set in the environment.
> kvm: -device virtio-vga-gl,id=vga,bus=pcie.0,addr=0x2: 'virtio-vga-gl' is not a valid device model name
> TASK ERROR: start failed: QEMU exited with code 1

jiangcuo, 2023-12-22:
new patch https://github.com/jiangcuo/qemu-server/commit/5e477dd2685bbe6491ec21b524923aef28f2d906

simper, 2023-12-23:
> new patch [jiangcuo/qemu-server@5e477dd](https://github.com/jiangcuo/qemu-server/commit/5e477dd2685bbe6491ec21b524923aef28f2d906)

With this applied on top, the VM can be started now, but spice display is black screen when connected with spice client, and on guest ubuntu serial terminal:

> root@ubuntu-server:~# glxinfo -B
> Error: unable to open display

> root@ubuntu-server:~# dmesg | grep gpu
> [    3.216984] [drm] pci: virtio-gpu-pci detected at 0000:00:02.0
> [    3.242059] [drm] Initialized virtio_gpu 0.1.0 0 for virtio0 on minor 0
> [    3.261456] virtio_gpu virtio0: [drm] drm_plane_enable_fb_damage_clips() not called
> [    3.624977] virtio_gpu virtio0: [drm] fb0: virtio_gpudrmfb frame buffer device

Still missing something?

simper, 2023-12-26:
I noticed the when "vga: virtio-gl" is set in vm conf, kvm will start with "-display egl-headless,gl=core", so even virgl is enabled, there is no way to render remote accessible desktop anymore, when I apply "-display spice-app,gl=core", the vm cannot be started:

> kvm: SPICE GL support is local-only for now and incompatible with -spice port/tls-port
so remote 3d accelerated desktop + sound is not possible on arm hosted VMs so far.

While only setting "args: -spice port=22222,addr=0.0.0.0,disable-ticketing=on" to start vm, then "remote-viewer --spice-debug spice://xxx.xxx.x.xxx:22222" to connect the VM remotely (with audio device added) did make sound work, although now with both sound and graphic processing burden fully on cpu, the cpu load is very high.

closed 2023-12-26 by simper