jiangcuo/pxvirt #15: VM crashing when console viewed

jfberry, 2023-12-21:
Randomly (but fairly often), the kvm process terminates when it is under inspection from the novnc viewer in the proxmox browser.
This is on an arm64 root server (ampere) in hetzner.

Installing symbols and tracing kvm in gdb indicates where the problem is:

```
Thread 15 "SPICE Worker" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xfffab37ea3e0 (LWP 238200)]
0x0000fffff7ef913c in pixman_composite_src_8888_8888_asm_neon ()
    at ../../pixman/pixman-arma64-neon-asm.S:1331
Downloading source file /build/pixman-phIJcH/pixman-0.42.2/build/pixman/../../pixman/pixman-arma64-neon-asm.S
1331    ../../pixman/pixman-arma64-neon-asm.S: Directory not empty.  
```

This has been reported upstream to debian maintainers, but reporting here in case anything can be done to influence this and for others.

Currently using a spice viewer for the console access and that hasn't shown any problems.
VMs stable once installed (console only being used for installation)

jiangcuo, 2023-12-22:
What hardware do you use, such as "lscpu"? And what is the version of pve, pveversion -v, what is the cmdline of the virtual machine? 
qm showcmd vmid --pretty

jiangcuo, 2023-12-22:
Reproduced in the latest pveversion

jiangcuo, 2023-12-22:
The above problems did not occur when using ramfb in the test, but the same problems occurred when using virtio-gpu and gl.

Before this problem is completely solved, it is recommended to use `vga: ramfb`

jfberry, 2023-12-22:
I confirm I was using virtio-gpu- since I regularly get this in the Ubuntu installer I will install some new machines with ramfb and confirm. Do you still want lscpu/pveversion now you have reproduced?

jfberry, 2023-12-22:
Unfortunately, ubuntu installer does not boot into install mode in ramfb gpu mode - initial boot loader and bios shows, but first reboot into main OS does not.

jiangcuo, 2023-12-22:
Thanks for the feedback, this issue should exist in pve8.X.

jiangcuo, 2023-12-26:
A new pixman version `libpixman-1-0=0.42.2-2` has been uploaded to repo,
do `apt update && apt install libpixman-1-0`  will  fix this issue (There were no same issues after the upgrade).

closed 2024-02-12 by jiangcuo