jiangcuo/pxvirt #16: Instructions for setting up PBS on LXC or VM

Anexgohan, 2023-12-21:
Hello,
first off amazing work, I managed to setup Proxmox on my new RPi5 withing an hour and have an lxc up and running, it was so easy and error free. thanks for this port.
secondly and my main query, I'm unable to setup proxmox backup server on a running lxc, is there any similar easy to follow guide/instructions for me to follow to set it up?

jiangcuo, 2023-12-24:
pbs need kernel 4k page size

Anexgohan, 2023-12-29:
> pbs need kernel 4k page size

thanks, i got PBS working on a VM on proxmox on RPi5, testing it now, although can't get PBS to "wipe the Disk" and "Initilize the disk" gives errors everytime "failed to execute sgdisk" something. same with adding a directory, same error. the only thing that works for now is "create:ZFS" from the PBS GUI, with that backups and restore work.
Any ideas why this sgdisk issue.

I'll post a how to later after testing and also I'm curious to get PBS working on LXC since it works on VM now.

Anexgohan, 2024-01-10:
update: Unable to passthrough disks to LXC but bind mounts of zfs from host to lxc works fine but the zfs ends up using the entire 8 gb RAM on the RPi5 so this is a non solution unless i can manually limit the zfs ram consumption.

PBS on the VM is working successfully for the past few days with disk passed through via simple USB port passthrough via GUI no configuration needed. Restore and backups work fine, note: cannot backup the vm itself (kinda obvious but just putting it here if I ever get the same stupid idea again :-)

will post the VM setup after a bit more testing.

Anexgohan, 2024-02-14:
Instructions to get ZFS and proxmox-backup-server working on a VM:
you need a VM running in proxmox ve.
tested for debian & ubuntu VM's

Get cloud image from here:
Debian:
- https://cloud.debian.org/images/cloud/
- https://jenkins.linuxcontainers.org/view/Images/job/image-debian/
- https://cdimage.debian.org/cdimage/cloud/

you need the genericcloud-arm64 image

Ubuntu:
- https://cloud-images.ubuntu.com/

you need the QCow2 UEFI/GPT Bootable disk image

# Setup Repos and other utils, (the utilities are optional):
```
apt update && \
apt install nano wget curl rsync bash-completion parted usbutils pciutils openssh-server ifupdown2 -y && \
nano /root/.bashrc && \
echo 'deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port'>/etc/apt/sources.list.d/pveport.list && \
echo 'deb https://mirrors.apqa.cn/proxmox/debian/pbs bookworm port'>/etc/apt/sources.list.d/pbs-port.list && \
curl https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o /etc/apt/trusted.gpg.d/pveport.gpg && \
apt update && apt full-upgrade -y
```

# add non free source:
```
echo "deb http://deb.debian.org/debian/ bookworm main contrib non-free non-free-firmware" | sudo tee -a /etc/apt/sources.list && \
echo "deb http://deb.debian.org/debian/ bookworm-updates main contrib non-free non-free-firmware" | sudo tee -a /etc/apt/sources.list && \
echo "deb http://security.debian.org/debian-security bookworm-security main contrib non-free non-free-firmware" | sudo tee -a /etc/apt/sources.list && \
echo "# see https://www.debian.org/doc/manuals/debian-reference/ch02.en.html#_updates_and_backports" | sudo tee -a /etc/apt/sources.list && \
echo "deb https://global.mirrors.apqa.cn/proxmox/debian/kernel sid port" | sudo tee -a /etc/apt/sources.list.d/kernel-port-sid.list
```

# Install kernel-header for Rasberry Pi 5:
```
apt update && apt search pve-kernel-6.1*
apt install pve-kernel-6.1-generic 
```

# Setup ZFS:
```
apt update && apt install dpkg-dev zfs-dkms zfsutils-linux -y
```
```
modprobe zfs
```

# Install proxmox-backup-server:
```
apt install proxmox-backup-server
```
# Disable the enterprise repo, since it will halt apt update :
```
sudo sed -i 's/^/#/' /etc/apt/sources.list.d/pbs-enterprise.list
```
# do an update and upgrade:
```
apt update && apt full-upgrade -y
```

# Remove broken and unnecessary leftovers
```
apt autoremove && \
apt autopurge && \
apt autoclean 
```

# Done 
log in to PBS :
https://ip-address-of-VM:8007/

Pass any disk by passing the entire USB port to the VM from the Proxmox VE GUI,
now you can use ZFS inside PBS and get the awesomeness that is ZFS and Deduplication

gmenezes-santos, 2024-03-26:
> Recently RPi 5 updates moved this file to:
> 
> `nano /boot/firmware/cmdline.txt`
> 
> old file location: (use which-ever works for you)
> 
> `nano /boot/cmdline.txt`
> 
> put the following parameters to the end of the line:
> 
> `cgroup_enable=cpuset cgroup_enable=memory cgroup_memory=1`

This is the fix for me, thanks!

Anexgohan, 2024-03-30:
> > Recently RPi 5 updates moved this file to:
> > `nano /boot/firmware/cmdline.txt`
> > old file location: (use which-ever works for you)
> > `nano /boot/cmdline.txt`
> > put the following parameters to the end of the line:
> > `cgroup_enable=cpuset cgroup_enable=memory cgroup_memory=1`
> 
> This is the fix for me, thanks!

glad it helped you.

closed 2024-04-13 by jiangcuo