jiangcuo/pxvirt #18: Kernel not built with br_netfilter

jfberry, 2023-12-22:
Using proxmox-port kernel 6.5.11-generic proxmox firewall cannot start

insmod br_netfilter fixes so have added to /etc/modules as a workaround but I am sure in the intel proxmox kernel this is not required

```
systemctl status pve-firewall
● pve-firewall.service - Proxmox VE firewall
     Loaded: loaded (/lib/systemd/system/pve-firewall.service; enabled; preset: enabled)
     Active: active (running) since Fri 2023-12-22 11:41:16 CET; 4min 38s ago
    Process: 4311 ExecStartPre=/usr/bin/update-alternatives --set ebtables /usr/sbin/ebtables-legacy (code=exited, status=0/SUCCESS)
    Process: 4313 ExecStartPre=/usr/bin/update-alternatives --set iptables /usr/sbin/iptables-legacy (code=exited, status=0/SUCCESS)
    Process: 4314 ExecStartPre=/usr/bin/update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy (code=exited, status=0/SUCCESS)
    Process: 4315 ExecStart=/usr/sbin/pve-firewall start (code=exited, status=0/SUCCESS)
   Main PID: 4319 (pve-firewall)
      Tasks: 1 (limit: 308045)
     Memory: 90.9M
        CPU: 2.325s
     CGroup: /system.slice/pve-firewall.service
             └─4319 pve-firewall

Dec 22 11:41:56 deviceLess pve-firewall[4319]: status update error: unable to open file '/proc/sys/net/bridge/bridge-nf-call-iptables' - No such file or direc>
Dec 22 11:42:06 deviceLess pve-firewall[4319]: status update error: unable to open file '/proc/sys/net/bridge/bridge-nf-call-iptables' - No such file or direc>
```

jiangcuo, 2023-12-22:
`br_netfilter` and `bridge` had builed as builtin on x86_64 kernel.
Can you help evaluate the following 2 solutions?
1. Compile the module as a builtin
2. I modified the pve-firewall service so that it loads br_netfilter modules when it runs

jfberry, 2023-12-22:
Either give me the right functional solution; whichever is simpler for your compatibility with upstream works for me

closed 2023-12-26 by jiangcuo