jiangcuo/pxvirt #23: problem in installing windows 11 arm on kunpeng 920

AxelPanda, 2024-01-06:
Unable to install Windows 11 arm version on Huawei Taishan 200-2280 (processor: Kunpeng 920 5220).
I followed the instructions [https://github.com/jiangcuo/Proxmox-Port/wiki/Windows-11-arm64](https://github.com/jiangcuo/Proxmox-Port/wiki/Windows-11-arm64) and created a VM in my PVE, but I was stuck in the installation progress.
Here is my VM's configuration file:
```
agent: 1
bios: ovmf
boot: order=scsi0;scsi2;net0;sata0
cores: 4
cpu: max
efidisk0: sata4t-zfs:vm-111-disk-0,efitype=4m,pre-enrolled-keys=1,size=64M
memory: 8192
meta: creation-qemu=7.0.0,ctime=1704454479
name: win11-arm-tmlp
net0: virtio=BC:24:11:AA:97:DB,bridge=vmbr4,firewall=1
numa: 0
ostype: l26
sata0: local:iso/virtio-win-0.1.240.iso,media=cdrom,size=612812K
scsi0: sata4t-zfs:vm-111-disk-1,size=32G
scsi2: local:iso/22621.1_PROFESSIONAL_ARM64_ZH-CN.ISO,media=cdrom,size=4601270K
scsihw: virtio-scsi-pci
smbios1: uuid=83f582a3-5d32-4d69-ac65-f3a5f8ce8799
sockets: 1
vga: ramfb
```

Here is the error page, which says this computer is not compatible with Windows 11:
[image]

I would like to know if there are any solutions to my situation. Thanks!

无法再华为泰山200-2280型服务器（处理器：鲲鹏920 5220）上安装Windows 11 arm版。
我按照wiki中的说明创建了虚拟机，但它卡在了安装过程中。
虚拟机配置文件和错误提示截图已在上文列出，请问是否有针对这一情况的解决办法？谢谢！

AxelPanda, 2024-01-06:
Problem solved. Use `Shift + F10` to call a cmd window, and start regedit.
Create an item named `LabConfig` under `HKLM\SYSTEM\Setup`.
Add two DWORD values named `BypassTPMCheck` and `BypassSecureBootCheck`, and set them to `1`.

问题已解决，在安装界面使用`Shift + F10`调出cmd窗口，使用regedit启动注册表编辑器。
在`HKLM\SYSTEM\Setup`下面新建一个名为`LabConfig`的项，
添加两个名为`BypassTPMCheck`和`BypassSecureBootCheck`的DWORD类型的值，并将值设置为`1`。

closed 2024-01-06 by AxelPanda

lelemka0, 2024-01-07:
This doesn't solve the problem, it just skips it.
There is a `secureboot` option in the `OVMF UEFI` settings, but it cannot be enabled. This feature is available in the x86 version of pve.
BTW I actually cannot install windows using the `virtio scsi dvd drive`, it fails during the boot phase saying boot device not found, so that I can only use USB mass storage device passthrough instead.

这并没有解决问题，只是跳过了问题。
在OVMF UEFI设置中有secureboot选项，但无法启用，该功能在x86版pve中是可用的。
另外事实上我无法使用 virtio scsi dvd 驱动器安装 Windows，它在启动阶段失败并提示未找到启动设备，因此我只能使用 USB 大容量存储设备直通作为替代。

nongze, 2024-01-30:
> Problem solved. Use to call a cmd window, and start regedit. Create an item named under . Add two DWORD values named and , and set them to .`Shift + F10``LabConfig``HKLM\SYSTEM\Setup``BypassTPMCheck``BypassSecureBootCheck``1`
> 
> 问题已解决，在安装界面使用调出cmd窗口，使用regedit启动注册表编辑器。 在下面新建一个名为的项， 添加两个名为和的DWORD类型的值，并将值设置为。`Shift + F10``HKLM\SYSTEM\Setup``LabConfig``BypassTPMCheck``BypassSecureBootCheck``1`

请问你使用的virtio-win-drivers是arm64版本的吗，还有可以提供一下arm64 win11的镜像吗，谢谢

nopmop, 2024-03-06:
Did anyone manage to load the Virtio drivers?