jiangcuo/pxvirt #25: secure boot not available

lelemka0, 2024-01-07:
There is an unavailable `secureboot` option in the `OVMF UEFI` settings, the `Attempt Secure Boot` option is grayed out and cannot be enabled.
[image]

related issue: #23

jiangcuo, 2024-01-07:
https://github.com/jiangcuo/pve-edk2-firmware/commit/a0b2e9cfedb77cf9458dfdf97705a4ca715da57c
https://github.com/jiangcuo/Proxmox-Arm64/issues/63

The fix will be done on the next version

closed 2024-01-22 by jiangcuo

lelemka0, 2024-03-18:
> @jiangcuo first of all thanks for your amazing work porting the proxmox! Can you please tell me how can I test the fix? Shall I build my own deb packages? Or I should wait till you release new updates for the deb packages? Maybe you can point to some howto on building debs from your repo's sources or perhaps small wiki page here for this would be helpful - other's like me can help with testing.

You can build [this](https://github.com/jiangcuo/qemu-server/tree/arm64) yourself to test the fix.
proxmox officially bumped the new version a few days ago, i think we will see an update soon.

lelemka0, 2024-03-18:
@bpopovych In fact, if you follow the Build instructions on the Readme of qemu-server, the compile product is `.deb`. Firmware using `make deb` is as the same.
If you don't want to compile on the PVE host, you can of course be performed on other hosts or in container.
You can compile in docker according to the following operation:
```
docker run -it debian:bookworm #For pve 8.x
apt update
apt install git wget sudo make rsync
wget http://download.proxmox.com/debian/proxmox-release-bookworm.gpg -qO /etc/apt/trusted.gpg.d/proxmox-release-bookworm.gpg
echo "deb http://download.proxmox.com/debian/pve bookworm pve-no-subscription" > /etc/apt/sources.list.d/pve-no-subscription.list

#!!! always compile the qemu-server first, compile script will install the necessary dependencies
git clone -b arm64 https://github.com/jiangcuo/qemu-server
cd ./qemu-server
bash ./autobuild.sh

#then build firmware
git clone -b arm64 https://github.com/jiangcuo/pve-edk2-firmware
cd ./pve-edk2-firmware
#necessary to fix repo path, otherwise the initialization of the submodule will fail
git submodule set-url edk2 https://git.proxmox.com/git/mirror_edk2.git
make deb
```

bpopovych, 2024-03-19:
@lelemka0 thanks for your help! I was able to run vm with Secure Boot enable/disable working UEFI feature using your instructions. However I gave up building pve-edk2-firmware on arm64 docker host - there should be some magic to disable X64/I32 architecture build for edk2-firmware - I was not able to find a way how to make it so I ended up building the pve-edk2-firmware on x64 docker host installing cross-compile toolchain. In fact the build for edk2-firmware is configured to produce x64, 32, arm/arm64 and riskV debs so I got the artefact I need. 
[image]
I'm attaching here two deb's I've build in case anyone else might need this before @jiangcuo will bump up new builds.

To enable fixed UEFI just unzip attached debs and type
`apt install ./qemu-server_8.0.10+pve1_arm64.deb ./pve-edk2-firmware-aarch64_4.2023.08-2_all.deb`
in the directory with debs.

[qemu+edk2.zip](https://github.com/jiangcuo/Proxmox-Port/files/14652527/qemu%2Bedk2.zip)