jiangcuo/pxvirt #26: How to passthrough Net card to VM on arm64

tomydec, 2024-01-09:
I passthrough a net card(intel 82599) to a vm on arm64 platform according to steps of x86's, have enabled ARM SMMU/IOMMU in BIOS, and passthrough it on grub, but when vm start with the pci of passthrough net card , it would be failed to start the vm. Cloud you help to give some guidance  how to do that? thanks

jiangcuo, 2024-01-09:
Can you share your config?
eg. vm , host ,grub,pveversion?

tomydec, 2024-01-09:
> Can you share your config? eg. vm , host ,grub,pveversion?

My steps:
1.Enable IOMMU in host server BIOS, and add“iommu.passthrough=1” after “quiet”，just like this “GRUB_CMDLINE_LINUX_DEFAULT="quiet iommu.passthrough=1"” and then “update-grub”；
2.add“vfio vfio_iommu_type1 vfio_pci vfio_virqfd” in /etc/modules；
3.add “blacklist ixgbe” in /etc/modprobe.d/pve-blacklist.conf，and then “update-initramfs -u -k all”；
4.reboot host server；
5.after rebooting，even install “vfio-pci” manually for Ether card，like this “echo vfio-pci >/sys/bus/pci/devices/0000\:14\:00.0/driver_override echo 0000:14:00.0 > /sys/bus/pci/drivers_probe”
6 and then add the pci to vm，but start with error “TASK ERROR: no such logical volume pve/data”

pveversion is 8.1.3

mrcheap1984, 2024-01-11:
Following this as well, I'm trying to pass a MediaTek wifi cards and getting bind errors.

`
fio-pci: probe of 0001:03:00.0 failed with error -22
`

jiangcuo, 2024-01-22:
https://github.com/jiangcuo/Proxmox-Port/wiki/Resource_PassThrough

closed 2024-01-22 by jiangcuo