jiangcuo/pxvirt #27: How to adjust the resolution of Windows 11 in the console

guoguo12343, 2024-01-10:
How to adjust the resolution of Windows 11 in the console？
[image]
I modified this resolution in the BIOS menu. I don't understand why there are so few resolution options.
[image]
The current resolution displayed in Windows has indeed increased.But the resolution of noVNC is still very low in full screen mode.
[image]

jiangcuo, 2024-01-12:
Use virtio-gpu when driver has installed
https://www.bilibili.com/video/BV1he41127P7/?spm_id_from=333.337.search-card.all.click

guoguo12343, 2024-01-15:
> Use virtio-gpu when driver has installed https://www.bilibili.com/video/BV1he41127P7/?spm_id_from=333.337.search-card.all.click

Thanks.That's too coincidental. This video was posted on the day I asked the question.

closed 2024-01-15 by guoguo12343