jiangcuo/pxvirt #28: installation on raspberry pi 5

perdicaro, 2024-01-10:
Hi everyone, I downloaded the latest available ISO and loaded it onto my SD card, but rpi5 can't find the boot and doesn't work, how can I install it?

cclngit, 2024-01-11:
What's not working? The installation of Proxmox-ARM or the installation of RaspberrypiOS on the RPi5? Because if it's the second problem, you're not in the right place. Please go to this forum: forums.raspberrypi.com. In my experience, installing RaspberrypiOS on RPis with Raspberrypi Imager is very simple. Maybe there's a problem with the SD card. You may have selected the wrong card, or the RPi5 may be defective. Good luck.

Anexgohan, 2024-01-11:
Read the wiki, it details how to get going, you need raspberry pi os lite (without gui, only ssh and cli), then add the sources.list and  apt install proxmox packages.

Read the wiki:
https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm

closed 2024-01-22 by jiangcuo