jiangcuo/pxvirt #30: Unable to connect to Internet on OrangePi 5 Plus (Host)

cclngit, 2024-01-12:
Hello, my main board is currently unable to connect to the internet. The OrangePi 5 Plus features two Ethernet ports, and I aim to designate them for WAN and LAN. To achieve this, I've created two bridges: vmbr0 for WAN (enP3p49s0) and vmbr1 for LAN (enP4p65s0), intending to implement a firewall. Currently, only VMs directly attached to vmbr0 can access the internet. However, as I am not proficient in configuring /etc/network/interfaces, I'm uncertain if there's an issue with my configuration. Could you please review it?

<details>
<summary>Info</summary>
<br>
OS: Proxmox VE 8.1.3+pve1 aarch64 <br>
Host: RK3588 OPi 5 Plus <br>
Kernel: 5.10.160-rockchip-rk3588 <br>
Uptime: 15 hours, 56 mins <br>
Packages: 926 (dpkg) <br>
Shell: bash 5.2.15 <br>
Terminal: /dev/pts/0 <br>
CPU: (8) @ 1.800GHz <br>
Memory: 1014MiB / 15970MiB
</details>

<details>
<summary>pveversion --verbose</summary>
<br>
proxmox-ve: 8.1.0 (running kernel: 5.10.160-rockchip-rk3588)<br>
pve-manager: 8.1.3+pve1 (running version: 8.1.3+pve1/26764642342c55bb)<br>
proxmox-kernel-helper: 8.1.0<br>
ceph: 17.2.6-pve1+3<br>
ceph-fuse: 17.2.6-pve1+3<br>
corosync: 3.1.7-pve3<br>
criu: 3.17.1-2<br>
dnsmasq: 2.89-1<br>
glusterfs-client: 10.3-5<br>
ifupdown: residual config<br>
ifupdown2: 3.2.0-1+pmx7<br>
libjs-extjs: 7.0.0-4<br>
libknet1: 1.28-pve1<br>
libproxmox-acme-perl: 1.5.0<br>
libproxmox-backup-qemu0: 1.4.0<br>
libproxmox-rs-perl: 0.3.1<br>
libpve-access-control: 8.0.7<br>
libpve-apiclient-perl: 3.3.1<br>
libpve-common-perl: 8.1.0-1<br>
libpve-guest-common-perl: 5.0.6<br>
libpve-http-server-perl: 5.0.4<br>
libpve-rs-perl: 0.8.7<br>
libpve-storage-perl: 8.0.5<br>
libspice-server1: 0.15.1-1<br>
lvm2: 2.03.16-2<br>
lxc-pve: 5.0.2-4<br>
lxcfs: 5.0.3-pve4<br>
novnc-pve: 1.4.0-3<br>
proxmox-backup-client: 3.1.2-1<br>
proxmox-backup-file-restore: 3.0.4-1<br>
proxmox-kernel-helper: 8.1.0<br>
proxmox-mail-forward: 0.2.0<br>
proxmox-mini-journalreader: 1.4.0<br>
proxmox-widget-toolkit: 4.1.3<br>
pve-cluster: 8.0.5<br>
pve-container: 5.0.8<br>
pve-docs: 8.1.3<br>
pve-edk2-firmware: not correctly installed<br>
pve-firewall: 5.0.3<br>
pve-firmware: not correctly installed<br>
pve-ha-manager: 4.0.3<br>
pve-i18n: 3.1.4<br>
pve-qemu-kvm: 8.1.2-4<br>
pve-xtermjs: 4.16.0-3<br>
qemu-server: 8.0.10+pve1<br>
smartmontools: 7.3-pve1
spiceterm: 3.3.0<br>
swtpm: 0.8.0+pve1<br>
vncterm: 1.8.0
</details>

> [!NOTE]
> pve-firmware: is correctly installed `Pin: version 3.20220526-rockchip`

<details>
<summary>/etc/network/interfaces</summary>
<br>
source /etc/network/interfaces.d/*

auto lo
iface lo inet loopback

iface enP4p65s0 inet manual

iface enP3p49s0 inet manual

auto vmbr0
iface vmbr0 inet static
	address 192.168.1.199/24
	gateway 192.168.1.1
	bridge-ports enP3p49s0
	bridge-stp off
	bridge-fd 0
	bridge-vlan-aware yes
	bridge-vids 2-4094

auto vmbr1
iface vmbr1 inet static
	address 192.168.1.200/24
	bridge-ports enP4p65s0
	bridge-stp off
	bridge-fd 0
	bridge-vlan-aware yes
	bridge-vids 2-4094
</details>

jiangcuo, 2024-01-12:
The configuration is fine, it may require you to check your network topology

cclngit, 2024-01-12:
Yes, it's my network topology, I managed to get it work.
The probleme came from the gateway, it's `192.168.1.254` instead of `192.168.1.1`.

closed 2024-01-12 by cclngit