jiangcuo/pxvirt #31: 能否提供 64k 内存页的内核 CONFIG_ARM64_64K_PAGES=y

PMExtra, 2024-01-16:
如题，这个项目兼容了很多信创平台的CPU，但是信创操作系统（包括银河麒麟在内）大多是64K内存页的，在当前的PVE上无法完美运行（个别应用对4K内存页兼容不佳会崩溃）。

所以希望有64K内存页的版本可供选择，理论上应该在kernel编译选项中去掉 `CONFIG_ARM64_4K_PAGES` 并设置 `CONFIG_ARM64_64K_PAGES=y` 就行。

tomydec, 2024-01-17:
you can complie the kernel according to https://github.com/jiangcuo/pve-port-kernel/tree/master, it's very easy. BTW, what's app has the compatible issue  you refered？Giva an example，I want to test it

jiangcuo, 2024-01-24:
64k pages are not compatible with current pve, and changing the kernel to a different page size means that some pve packages need to be rebuilt. Therefore, I will not release a 64k kernel. If you want to use a 64k kernel, you can build it yourself. You can also buy the corresponding technical services.

closed 2024-02-12 by jiangcuo