jiangcuo/pxvirt #36: riscv64 port: cannot install on Debian sid

xen0n, 2024-01-21:
When following the installation steps on a Debian sid system, I got this:

```
$ sudo apt install proxmox-ve postfix open-iscsi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libpve-rs-perl : Depends: perlapi-5.36.0 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.36.0 but it is not installable
 librados2-perl : Depends: perlapi-5.36.0 but it is not installable
 pve-cluster : Depends: perlapi-5.36.0 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Seems the packaging scheme has changed Debian-side?

liberodark, 2024-01-24:
That have been work for me.

liberodark, 2024-02-08:
Also you can setup this version for install PVE : https://snapshot.debian.org/archive/debian/?year=2023&month=10

Evsio0n, 2024-12-10:
Issue seems same to https://forum.proxmox.com/threads/proxmox-completely-broken.147080/

Looks like it's an debian sid repository issue.

jiangcuo, 2025-04-23:
I have provided a package containing proxmox-ve-riscv rootfs. Since riscv currently does not have hardware for H extension, only lxc is available. This package is only for those who are interested to test

https://mirrors.lierfang.com/proxmox/lxcs/proxmox-ve-riscv64-rootfs.tar.gz

closed 2025-04-23 by jiangcuo