jiangcuo/pxvirt #38: 飞腾S2500双路服务器上启动失败

123shang60, 2024-01-23:
尝试在清华同方超强F520-M1+的机器上使用 ISO 镜像安装Proxmox VE 8.1-3 版本，会有如下的问题：

当 BIOS 中 Socket 数量配置为 1 时，可以正常引导 ISO 镜像，可以正常进入安装程序，并正常安装；安装后，也可以正常进入系统；但是当 BIOS 中 Socket 数量配置为 2 时，则在引导阶段会卡死，无法正常启动。

Socket 配置为 2，启动异常时的串口输出日志为：

```
FSOpen: Open '\EFI\proxmox\grubaa64.efi' Success
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x8,0x0)/Pci(0x0,0x0)/Ctrl(0x0)/Scsi(0x0,0x0)/HD(2,GPT,408FCEDC-8FCA-4BCE-8755-1681DD76A161,0x800,0x200000)/\EFI\proxmox\grubaa64.efi.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B F2FB8440
xfliu: In Image.c, line 624. 
Loading driver at 0x000EC6AF000 EntryPoint=0x000EC6B0000 
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF EF653298
ProtectUefiImageCommon - 0xF2FB8440
  - 0x00000000EC6AF000 - 0x0000000000428000
InstallProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA FBFFF658
[Security] 3rd party image[0] can be loaded after EndOfDxe: MemoryMapped(0x2,0xE9FF8000,0xEC6AEA00).
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B F2FB8AC0
xfliu: In Image.c, line 624. 
Loading driver at 0x000E51B0000 EntryPoint=0x000E6B82AB8 
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF ED565418
ProtectUefiImageCommon - 0xF2FB8AC0
  - 0x00000000E51B0000 - 0x00000000027C0000
XhcClearBiosOwnership: called to clear BIOS ownership
XhcClearBiosOwnership: called to clear BIOS ownership
SetUefiImageMemoryAttributes - 0x00000000F2F30000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F2E90000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F2DE0000 - 0x0000000000050000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F2CF0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F2C50000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F2B30000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F06A0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F0600000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F04B0000 - 0x0000000000040000 (0x0000000000000008)
SetUefiImageMemoryAttributes - 0x00000000F03E0000 - 0x0000000000040000 (0x0000000000000008)
NOTICE:  Standard Service Call: 0x84000000 
NOTICE:  return value: 65536
NOTICE:  Standard Service Call: 0x84000006 
NOTICE:  return value: 4294967295
NOTICE:  Standard Service Call: 0x8400000a 
NOTICE:  return value: -1
NOTICE:  Standard Service Call: 0x8400000a 
NOTICE:  return value: 0
NOTICE:  Standard Service Call: 0x8400000a 
NOTICE:  return value: -1
NOTICE:  Standard Service Call: 0x8400000a 
NOTICE:  return value: -1
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0x1
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 1
NOTICE:  ft2500 pwr domain on 0x1
NOTICEN:O  IreEt:u r nf tv2alue:w0
omain_on_finish
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0x2
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 2
NOTICE:  ft2500 pwr domain on 0x2
NOTICE:  return vEl e: 0
0_pwr_domain_on_finish
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0x3
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 3
NOTICE:  ft2500 pwr domain on 0x3
NOTICE:  ret rn value: _d
                         main_on_finish
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0x100
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 4
NOTICE:  ft2500 pwr domain on 0x100
ain_on_finisht2n 0apwe_ o
[... 301 lines cut ...]
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 62
NOTICE:  ft2500 pwr domain on 0xf02
NONITICE:  fu25 0_pwe_domain_on_finish
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0xf03
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 63
NOTICE:  ft2500 pwr domain on 0xf03
NOTOTECE:reft25 v_pwr_d0m
                         in_on_finish
NOTICE:  Standard Service Call: 0xc4000003 
NOTICE:  PSCI CPU ON entryponit: 0xb9eaf2c8, context_id: 0x0, target_cpu: 0x10000
NOTICE:  psci cpu on start-pwr_domain_on,target_cpuidx: 64
NOTICE:  ft2500 pwr domain on 0x10000
NOTICE:  return value: 0
```

机器的关键硬件配置大致如下：

CPU - 两颗 FT-S2500 
内存 - 8 * 32G，共 256G 内存

机器目前安装其他国产化、内核版本为 4.10.x 或者 5.10.x 的 linux 系统可以正常启动，无明显的硬件兼容性问题；因为我本人对 PVE的内核不太了解，想咨询两个问题：

1. 这种无法启动，大概是什么会是什么原因导致的？是否是内核的问题？
2. PVE 8.1 版本是否可以更换为低版本的内核，比如 5.10.x 等版本？

jiangcuo, 2024-01-24:
You build kernel from `pve-port-kernel` if default kernel has some issue.

Johnson-W, 2024-03-22:
楼主有解决办法了没呢？我也遇到一样的问题，CPU也是两颗 FT-S2500
pve、debian9到debian12都安装不了，开了grub debug后显示全部卡在EFI stub:Exiting boot seruices
但Kylin-Server-V10-SP3-General-Release-2303-ARM64.iso却可以正常安装没有问题
理论上debian11是5.10的内核，但几个小版本都试过了也是不行，还是卡在EFI stub:Exiting boot seruices
感觉是debian的iso哪里有不兼容的地方

123shang60, 2024-03-22:
之前做过一些研究，大致有一些思路：

大概率问题出现在 FTS2500 需要特殊的 GIC 处理逻辑，具体可以参考 `openEuler-kernel` 中的一些 `commit`

[16dcf922d66519941ce6f7f95d4c678cf4aadadb](https://gitee.com/openeuler/kernel/commit/16dcf922d66519941ce6f7f95d4c678cf4aadadb) 

目前我经过测试，找到理论可行的办法是：

1. 先修改FTS2500 BIOS 配置，开启兼容模式，CPU 数量改为 1 ，这个时候不能识别到完整的 PCIE ，但是机器可以正常引导、安装 PVE
2. 基于 PVE 的 `kernel` 代码，将上述在 `openEuler-kernel` 中的代码迁移过去，重新编译内核
3. 在 FTS2500 的 PVE 环境上安装，安装后重启，修改 BIOS 位兼容模式关闭，CPU 数量 2 ，理论上就可以重新启动

但是因为工作关系，我暂时还没时间完整的对上述方案进行验证

Johnson-W, 2024-03-22:
感谢，下周找时间拆掉一颗U试试，测试的机器用的中电科技BIOS，CPU设置项完全不能调整，太多坑了

jiangcuo, 2024-04-13:
最新的的8.1.7的iso中，集成了欧拉的内核，可以试试。安装和启动都使用欧拉的内核，

Johnson-W, 2024-04-16:
> 最新的的8.1.7的iso中，集成了欧拉的内核，可以试试。安装和启动都使用欧拉的内核，

机器配的是中电科技BIOS和双路飞腾S2500，用了8.1.7的ISO仍然会卡在EFI stub:Exiting boot seruices。
这个机器目前测试唯一能正常安装的只有麒麟V10系统，内核版本是kernel-4.19.90-25.10.v2101.ky10.aarch64或者kernel-4.19.90-52.22.v2207.ky10.aarch64
BIOS信息显示厂商中电科技，UEFI2.7;PI1.6，固件版本1.0.9，建立时间10/27/2021
可能飞腾自己的BIOS兼容性会好些，中电科技的BIOS目前测试下来属实拉胯

XizumiK, 2024-09-29:
问过搞BIOS的朋友，飞腾S2500双路中断设计有问题，需要打补丁的特殊内核

飞腾应该是有内核补丁，可以发邮件试试但他们不一定给

还有一个办法是直接关闭一路CPU，但我手上这台联想的机器关掉的那路CPU上挂着USB控制器，关掉以后所有USB都用不了...

jiangcuo, 2024-09-29:
欧拉如果不能用，那开源的应该都不能吧。

XizumiK, 2024-09-29:
> 欧拉如果不能用，那开源的应该都不能吧。

是的，只要飞腾不给打了中断补丁的专用内核，在双路S2500上都不能启动。现在只有银河麒麟这个商业发行版自带这个内核能够正常启动。

飞腾S5000貌似已经修复了这个问题，双路S5000跑Ubuntu挺稳的。

所以这个Issue其实已经可以关了，飞腾看起来不打算解决这个问题，文档和补丁也不开放。其实所谓的国产“信创”全都是这毛病，东西做得烂就算了，文档和资料还给你藏着掖着。。。

jiangcuo, 2025-04-23:
在最新的内核中已解决，支持s2500和s5000c

closed 2025-04-23 by jiangcuo