jiangcuo/pxvirt #39: Raspberry Pi 4 - Proxmox - ceph

mikev1963, 2024-01-23:
when trying to install ceph via proxmox I get the following error:

Jan 23 15:59:47 node12 ceph-mon[13545]: 2024-01-23T15:59:47.249-0500 7f66fdc280 -1 mon.node12@2(probing) e0 handle_probe require release 18 > 16, or missing features (have 4540138314316775423, required 0, missing 0)

It appears the Raspberry Pi needs version 18 rather than 16.  Is there a way to upgrade this myself?

jiangcuo, 2024-01-24:
Default ceph version is 17.If you want install ceps-reef,add  `deb https://mirrors.apqa.cn/proxmox/debian/pve bookworm ceph-reef` in to sources.list

Remember ,Don't install ceph by webui or pveceph

barlasdnz20, 2024-02-12:
Hi everyone, I'm a novice user and I'm trying to set up a container disk using Ceph. I've never used Ceph before and I'm encountering some errors during the installation process. Would anyone be able to provide me with step-by-step instructions on how to install Ceph?"

I'm sorry for my bad english. :)

jiangcuo, 2024-03-15:
There was no update, so the issue was closed

closed 2024-03-15 by jiangcuo