jiangcuo/pxvirt #41: invalid ipset name

liberodark, 2024-01-26:
Hi,

I currently have a PVE 8.1.x on RISC-V
And I'm stuck on a small problem defining via ipset.
I can't use my firewall as I would like.My configuration works on a classic PVE.
But here I tell myself that I probably forgot a defconfig side for the kernel.
Do you have any ideas or recommendations on this?

Best Regards

[image]

```
Jan 26 12:15:58 sv1 pve-firewall[4109]: /etc/pve/firewall/cluster.fw (line 25) - errors in rule parameters: IN HTTPS(ACCEPT) -source +dc/whitelist -log nolog
Jan 26 12:15:58 sv1 pve-firewall[4109]:   source: invalid ipset name '+dc/whitelist'
Jan 26 12:16:08 sv1 pve-firewall[4109]: /etc/pve/firewall/cluster.fw (line 21) - errors in rule parameters: IN Ping(ACCEPT) -source +dc/whitelist -log nolog
Jan 26 12:16:08 sv1 pve-firewall[4109]:   source: invalid ipset name '+dc/whitelist'
Jan 26 12:16:08 sv1 pve-firewall[4109]: /etc/pve/firewall/cluster.fw (line 22) - errors in rule parameters: IN ACCEPT -source +dc/whitelist -p tcp -dport 800>
Jan 26 12:16:08 sv1 pve-firewall[4109]:   source: invalid ipset name '+dc/whitelist'
Jan 26 12:16:08 sv1 pve-firewall[4109]: /etc/pve/firewall/cluster.fw (line 23) - errors in rule parameters: IN SSH(ACCEPT) -source +dc/whitelist -log nolog
Jan 26 12:16:08 sv1 pve-firewall[4109]:   source: invalid ipset name '+dc/whitelist'
Jan 26 12:16:08 sv1 pve-firewall[4109]: /etc/pve/firewall/cluster.fw (line 25) - errors in rule parameters: IN HTTPS(ACCEPT) -source +dc/whitelist -log nolog
Jan 26 12:16:08 sv1 pve-firewall[4109]:   source: invalid ipset name '+dc/whitelist'
```

jiangcuo, 2024-01-26:
Enable ipset in kernel  config

liberodark, 2024-01-27:
Yes that have been made same as official PVE.
But that not fix my issue.

liberodark, 2024-01-30:
The  issue is probably related to : https://lists.proxmox.com/pipermail/pve-devel/2024-January/061388.html
I need to check that.

OfficialMuffin, 2024-02-03:
I have `pve-firewall: 5.0.3` installed. I am still getting the same error. I am using the CIDR notation in the source box.

liberodark, 2024-02-08:
What is your hardware ?

closed 2025-07-20 by jiangcuo