jiangcuo/pxvirt #44: No user/password defined whilke installing via repo

Tacoos33, 2024-01-28:
**Describe the bug**
I installed on a raspberry pi 5 as described in the wiki, using the repo method.
After install, i can connect to http://myraspberrypi:8006 but i cannot login because during install i was not asked to define a password for the root user, and the default login for my pi does not work
Or did i miss something in the wiki?

jackydec, 2024-01-28:
had the same problem installing on an odroid m1s. 
what you need to do is :
- on your raspberry pi  command line : login as root via : sudo su -
- after this, you should be root user.
- then type in : passwd    -> it will ask you to set a new password for your root user (and this is the same to login to proxmox)

Tacoos33, 2024-01-28:
Thanks, that did the trick indeed

jiangcuo, 2024-01-31:
This is not proxmox-ve issue,close it

closed 2024-01-31 by jiangcuo