jiangcuo/pxvirt #45: Status of RISC-V support

dfsek, 2024-01-28:
Hi! I am considering getting a [Milk-V Pioneer](https://milkv.io/pioneer) board to add RISC-V to my homelab. I was wondering

 * What the maturity of the RISC-V port is, is it stable/usable? I have seen that development of it is not as active as the other platforms
 * Whether anyone has tested on the Milk-V Pioneer specifically. It seems to be the highest-performance, and only "server-class" RISC-V platform currently available.

Thank you! And thank you for your effort on these ports, I am already using the ARM versions in my lab, I really appreciate all your work! :)

smiba, 2024-01-28:
Wondering the same, been watching this repo for a while as I have my eyes on some RISC-V hardware

liberodark, 2024-01-29:
I im on RISC-V : 

[image]
Im go to use Milk-V Hardware too.

jiangcuo, 2024-01-31:
Promxox VE for riscv64 which based debian sid has  been released 
You can use [debian-sid(202310)](https://snapshot.debian.org/archive/debian/?year=2023&month=10) and install it .

closed 2025-04-23 by jiangcuo