jiangcuo/pxvirt #47: Your ISOs and rockchip

brightplastik, 2024-02-02:
Hello jiangcuo. I'm amazed with your work, and wish I was so clever.

Sorry to bother you, but since weeks I'm trying to install your port to a friendlyelec nanopc t6, with a copy of edk2 uefi on the NOR chip, and all my attempts fail miserably, both with ACPI and device tree setting.

Can you give me a hint? Do you think it is possible?

I have an idea it has to do with the load of custom rockchip DTB, but I'm too noob to have a precise clue.

Any help would be highly appreciated.

The only image I was able to install on UEFI is debian from amazingfate github.

jiangcuo, 2024-02-04:
Please try installing from debian

ProKn1fe, 2024-02-08:
Board should be supported by kernel which is most likely not.

brightplastik, 2024-02-08:
@ProKn1fe do you mean that it should be supported in a ideal life but in reality it isn't? Yes, exactly. I'm in contact with the EDK2 porting developers, and wish this soc to get some proper IOMMU passthrough.
I'll return here with news, if I get anything to work.
Meanwhile, any suggestion is welcome!

brightplastik, 2024-02-28:
OK, so time for an update.
In order to boot the ISO image on the nanopc-T6 I had to be cautious about the medium I was using and the UEFI settings on the edk2 firmware. It is doable, and I was only able to boot in these conditions:
- 6.1 kernel and downward
- a USB medium, either via Ventoy (latest is important!) or Rufus (sd card did not work)
- ACPI setting in UEFI
- either 8.0-1 or 8.1-3, but in the latest there's more chance to work with passthrough maybe?

This is the serial log until second debug prompt: https://pastebin.com/2AqYRcym

Since many arm PCs come with MMC, I tried to change the installer as documented [here](https://ibug.io/blog/2022/03/install-proxmox-ve-emmc/) but till now I failed, as in the installer the MMC does not show at all, unfortunately.
In the specific case of your conversion for ARM64 boards, I think allowing the possibility to install on MMC would be interesting to many of the users.
Therefore, I suggest to modify `pve-installer.squashfs` to insert a modified copy of `/usr/share/perl5/Proxmox/Sys/Block.pm` with the following:

>  } elsif ($dev =~ m|^/dev/[^/]+/hd[a-z]$|) {
     return "${dev}$partnum";
 } elsif ($dev =~ m|^/dev/nvme\d+n\d+$|) {
     return "${dev}p$partnum";
 } elsif ($dev =~ m|^/dev/mmcblk\d+$|) {
     return "${dev}p$partnum";
 } else {
     die "unable to get device for partition $partnum on device $dev\n";
 }
so that the ISO can possibly find and select that medium during install.
Maybe it is not so straightforward.
In that case, I apologize to simplify things.

For example of ipersemplification, if I run `lsblk` the only recognized devices are 
> loop0 
loop1
sda
 -sda1
 -sda2
So it might be more complex than this.
Hope this comment can be useful.
Take care geeks!

jiangcuo, 2024-02-29:
The kernel in the ISO is the mainline kernel,  it may fail to start without the corresponding patch If it is a soc device.
I only have a standard arm64 device, not a development board, so I can't test it.
If you are willing to test, I can rebuild a kernel for the rk device.
https://github.com/jiangcuo/Proxmox-Port/wiki/Resource_PassThrough

brightplastik, 2024-02-29:
The T6 is not productive yet. I took the occasion to learn new things I the meanwhile, so I can reinstall the OS on the box many times. I have snapshots on timeshift, to revert eventually. 
I'm more than happy to test things @jiangcuo, if you want to recompile 🙏

jiangcuo, 2024-02-29:
New kernel has been upload
- Kernel download link 
https://mirrors.apqa.cn/proxmox/debian/kernel/dists/sid/test/binary-arm64/
- Kernel source
https://github.com/jiangcuo/pve-port-kernel/tree/rockchip 
You can use Debian and then install deb. Try booting with a new kernel.

brightplastik, 2024-03-01:
>Try booting with a new kernel.
> 

Tried that. Unfortunately out of the easy route  @amazingfate is doing with the dtb injection the system stalls and there is no OS.
I tried to inject it via grub, and there is some sort of improvement (IP assigned and active mouse and keyboard), but still I don't reach a desktop screen.
If needed, I can provide a serial log, opening the box and attaching a UART.
I'm so sorry to be unable to support with some proper linux knowledge and proactivity.

jiangcuo, 2024-03-02:
The logs show that rootdisk is not found. 
Would it be possible to  give the following command output?
- `blkid`
- `ls /sys/bus/platform/devices/* -l`
- `ls /sys/bus/platform/devices/*/driver -l`

jiangcuo, 2024-03-02:
`dmesg`

brightplastik, 2024-03-02:
```
```
```
```
```
```
```
```

brightplastik, 2024-03-02:
Hello Foxi (right?). As usual, I want to tell you I appreciate a lot your support. 
None of this testing would be possible without your knowledge. If we ever get to a result, please tell me how I can give back to you.
Regarding what you asked, The following commands are:

**with your test 6.1.43 kernel that does not boot, launched in initramfs:**
```
(initramfs) blkid
```
no result

```
(initramfs) ls /sys/bus/platform/devices/* -l
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/Fixed MDIO bus.0 -> ../../../devices/platform/Fixed MDIO bus.0
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/HISI0031:00 -> ../../../devices/platform/HISI0031:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PNP0D10:00 -> ../../../devices/platform/PNP0D10:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PNP0D10:01 -> ../../../devices/platform/PNP0D10:01
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PNP0D20:00 -> ../../../devices/platform/PNP0D20:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PNP0D20:01 -> ../../../devices/platform/PNP0D20:01
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PRP0001:00 -> ../../../devices/platform/PRP0001:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PRP0001:01 -> ../../../devices/platform/PRP0001:01
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/PRP0001:02 -> ../../../devices/platform/PRP0001:02
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP0D40:00 -> ../../../devices/platform/RKCP0D40:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:00 -> ../../../devices/platform/RKCP3001:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:01 -> ../../../devices/platform/RKCP3001:01
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:02 -> ../../../devices/platform/RKCP3001:02
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:03 -> ../../../devices/platform/RKCP3001:03
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:04 -> ../../../devices/platform/RKCP3001:04
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:05 -> ../../../devices/platform/RKCP3001:05
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:06 -> ../../../devices/platform/RKCP3001:06
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3001:07 -> ../../../devices/platform/RKCP3001:07
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3002:00 -> ../../../devices/platform/RKCP3002:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3002:01 -> ../../../devices/platform/RKCP3002:01
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3002:02 -> ../../../devices/platform/RKCP3002:02
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3002:03 -> ../../../devices/platform/RKCP3002:03
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3002:04 -> ../../../devices/platform/RKCP3002:04
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCP3003:00 -> ../../../devices/platform/RKCP3003:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/RKCPFE2C:00 -> ../../../devices/platform/RKCPFE2C:00
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/dmabuf -> ../../../devices/platform/dmabuf
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/efivars.0 -> ../../../devices/platform/efivars.0
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/mtd_vendor_storage -> ../../../devices/platform/mtd_vendor_storage
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/psci-cpuidle -> ../../../devices/platform/psci-cpuidle
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/reg-dummy -> ../../../devices/platform/reg-dummy
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/regulatory.0 -> ../../../devices/platform/regulatory.0
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/rtc-efi.0 -> ../../../devices/platform/rtc-efi.0
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/serial8250 -> ../../../devices/platform/serial8250
lrwxrwxrwx    1 root     0                0 Jan  1 00:00 /sys/bus/platform/devices/snd-soc-dummy -> ../../../devices/platform/snd-soc-dummy
(initramfs)
```
```
(initramfs) ls /sys/bus/platform/devices/*/driver -l
lrwxrwxrwx    1 root     0                0 Jan  1 00:22 /sys/bus/platform/devices/HISI0031:00/driver -> ../../../bus/platform/drivers/dw-apb-uart
lrwxrwxrwx    1 root     0                0 Jan  1 00:22 /sys/bus/platform/devices/PNP0D10:00/driver -> ../../../bus/platform/drivers/xhci-hcd
[... cut ...]
[   52.044985] input: Multimedia keyboard System Control as /devices/platform/PNP0D10:00/usb3/3-1/3-1.4/3-1.4:1.1/0003:060B:2081.0003/input/input3
[   52.045135] hid-generic 0003:060B:2081.0003: input,hidraw2: USB HID v1.11 Device [Multimedia keyboard] on usb-PNP0D10:00-1.4/input1
[   53.521547] vendor storage:20190527 ret = -1
[   60.624991] platform mtd_vendor_storage: deferred probe pending
(initramfs)
```
**the same commands launched from terminal in amazingfate debian running 5.10.160-legacy-rk35xx kernel:**

```
```

jiangcuo, 2024-03-03:
The dmesg-out and devices not show any mmcblk.
What's your dtbs file? I think we should use the dtb integrated with the kernel package.

kernel has built-in  mmc driver
```
root@arm-builder3:~/pve-port-kernel/linux# cat modules.builtin|grep mmc|grep -E "sdh|rock"
kernel/drivers/phy/rockchip/phy-rockchip-emmc.ko
kernel/drivers/soc/rockchip/sdmmc_vendor_storage.ko
kernel/drivers/mmc/host/sdhci.ko
kernel/drivers/mmc/host/dw_mmc-rockchip.ko
kernel/drivers/mmc/host/sdhci-pltfm.ko
kernel/drivers/mmc/host/sdhci-of-arasan.ko
kernel/drivers/mmc/host/sdhci-of-dwcmshc.ko
```
I can also use his kernel tree （https://gitlab.collabora.com/hardware-enablement/rockchip-3588/linux/-/tree/rk3588-v6.6?ref_type=heads）to build, but the device information cannot be correctly identified using the official kernel. It should be that the dtb is incorrect.

jiangcuo, 2024-03-03:
I compared the two kernels and found that Collabora's kernel has dts for many models, so I recompiled a new package based on Collabora's kernel （6.7）.
https://mirrors.apqa.cn/proxmox/debian/kernel/dists/sid/test/binary-arm64/

brightplastik, 2024-03-03:
I will try it ASAP.
I hope I won't confuse my lack of ability with a bug, and slow down development.
Finger crossed this might be the good way to take.
I'll do my best.
Meanwhile, please tell me if this commands are needed to debug things when the kernel is installed:
>

jiangcuo, 2024-03-05:
The new kernel is Collabora’s kernel,I think there should be no problem.

brightplastik, 2024-03-06:
> The new kernel is Collabora’s kernel,I think there should be no problem.
> 

Worse than ever, unfortunately.

On edk2, with ACPI selected, as soon as it boots it stalls.

I need the DTB file related to this kernel, maybe, and try to override. The DTB of the legacy rockchip 5.10.160 kernel does not work with this new kernel.

I can show a log, but there's nothing to show, this time.

jiangcuo, 2024-03-08:
Dtbs are located at "/boot/dtbs/{kernel}/"

brightplastik, 2024-03-10:
> Dtbs are located at "/boot/dtbs/{kernel}/"

Ok, I changed the path in grub and now this is what comes in the serial log...

```
[    2.160050] Buffer I/O error on dev mmcblk0p1, logical block 76685, async page read
/dev/mmcblk0p2: clean, 168850/15089664 files, 2809818/60338432 blocks
[    4.884363] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007
[    4.885042] CPU: 7 PID: 1 Comm: systemd Not tainted 6.7.0-rockchip-gb07290444a7f-dirty #1
[    4.885761] Hardware name: FriendlyElec NanoPC T6/NanoPC T6, BIOS 4af3a03 01/20/2024
[    4.886438] Call trace:
[    4.886656]  dump_backtrace+0xa0/0x128
[    4.886994]  show_stack+0x20/0x38
[    4.887288]  dump_stack_lvl+0x78/0xc8
[    4.887617]  dump_stack+0x18/0x28
[    4.887912]  panic+0x3b8/0x400
[    4.888187]  do_exit+0x818/0x918
[    4.888479]  do_group_exit+0x3c/0xa8
[    4.888799]  get_signal+0xa40/0xa48
[    4.889110]  do_notify_resume+0x290/0x13e0
[    4.889472]  el0_ia+0xc0/0xc8
[    4.889740]  el0t_64_sync_handler+0xd0/0x130
[    4.890120]  el0t_64_sync+0x188/0x190
[    4.890446] SMP: stopping secondary CPUs
[    4.890910] Kernel Offset: disabled
[    4.891217] CPU features: 0x1,80000000,70028146,2101720b
[    4.891685] Memory Limit: none
[    4.918244] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007 ]---
``` 

Is it my mistake? 
I think I managed to override the DTB provided by your kernel, but still, it produces a kernel panic.
Maybe Collabora's kernel is not meant to work outside of U-boot environment?

jiangcuo, 2024-03-10:
> Ok, I changed the path in grub and now this is what comes in the serial log...
>
> ```
> [    2.160050] Buffer I/O error on dev mmcblk0p1, logical block 76685, async page read
> /dev/mmcblk0p2: clean, [...]

I don't have a development board, so I can't debug, if the debian kernel works, then it is recommended to install debian and use proxmox-ve

jiangcuo, 2024-04-04:
New ISO release, updated PVE to 8.1.7, support for orange pi to start ISO via UEFI. Download link https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.1-7_arm64.iso
Follow wiki to setup
https://github.com/jiangcuo/Proxmox-Port/wiki/Rockchip-RK3588-UEFI-Setup

876873549, 2024-04-06:
[image]
proxmox-ve: 8.1.0 (running kernel: 6.1.43-rockchip)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.73-generic: 6.1.73-1
pve-kernel-6.1-rockchip: 6.1.43-1
pve-kernel-6.1.43-rockchip: 6.1.43-1
pve-kernel-5.15.145-generic: 5.15.145-1
pve-kernel-5.10.0-openeuler: 5.10.0-1
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: not correctly installed
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.6
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: 4.2023.08-4
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port2
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.3-pve1

安装了WIFI网卡MA210，我还是不知道怎么使用pcie-passthrough

876873549, 2024-04-06:
> 新ISO版本，更新PVE至8.1.7，支持orange pi通过UEFI启动ISO。下载链接https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.1-7_arm64.iso 按照wiki设置 [https://github.com/jiancuo/Proxmox-Port/wiki/Rockchip-RK3588-UEFI -设置](https://github.com/jiangcuo/Proxmox-Port/wiki/Rockchip-RK3588-UEFI-Setup)

[image]
proxmox-ve: 8.1.0 (running kernel: 6.1.43-rockchip)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.73-generic: 6.1.73-1
pve-kernel-6.1-rockchip: 6.1.43-1
pve-kernel-6.1.43-rockchip: 6.1.43-1
pve-kernel-5.15.145-generic: 5.15.145-1
pve-kernel-5.10.0-openeuler: 5.10.0-1
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: not correctly installed
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.6
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: 4.2023.08-4
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port2
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.3-pve1

安装了wifi网卡AX210.  但我还是不知道怎么pcie-passthrough

876873549, 2024-04-06:
root@pve:/# mount -t cifs //192.168.0.65/Z /mnt/a/ -o username=p,password=p
mount error: cifs filesystem not supported by the system
mount error(19): No such device
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
root@pve:/#      
挂载失败

brightplastik, 2024-04-06:
> > 
> [image]
> 
As far as I understood IOMMU is still a big work in progress, and passes by a modified kernel with some magic infused in it. @jiangcuo tried sharing a modified kernel for nanopc t6 previously, but the kernel only works in uboot env, and it failed in my edk2 environment.

brightplastik, 2024-04-06:
> New ISO release, updated PVE to 8.1.7, support for orange pi to start ISO via UEFI. Download link https://mirrors.apqa.cn/proxmox/isos/proxmox-ve_8.1-7_arm64.iso
> Follow wiki to setup
> https://github.com/jiangcuo/Proxmox-Port/wiki/Rockchip-RK3588-UEFI-Setup

Hello Foxi. I guess this will not be useful (yet?) for nanopc t6 platform.
Is there anything specific I can try, with my edk2 board? I lack knowledge, but willing to try things, if available.

jiangcuo, 2024-04-13:
The new ISO already supports the full series of orangpi5. If I have other rockchip adaptations, I'll try to integrate the kernels that work for them.A temporary way is to use ACPI mode, which will lose most of the functionality, such as emmc, npu, etc.I'm going to close the issue because I can't help it.

closed 2024-04-13 by jiangcuo

LabuzhskiyAleksandr, 2024-05-27:
Hi, Guys! I think, it's works on the Nanopc-t6:
[Nanopc-T6 с Proxmox 8 _ Песочница _ Хабр.pdf](https://github.com/jiangcuo/Proxmox-Port/files/15461662/Nanopc-T6.Proxmox.8._._.pdf)
[image]

brightplastik, 2024-05-29:
> Hi, Guys! I think, it's works on the Nanopc-t6:
>

This is fantastic news Alexander! 
Did you manage to passthrough the pci devices?