jiangcuo/pxvirt #50: NO DISPLAY ON CONSOLE

muko82, 2024-02-11:
**Describe the bug**
After installing proxmox is not possible to access VM console display getting "guest has not initialized display yet"

**To Reproduce**
Steps to reproduce the behavior:
1. Install proxmox
2. Try to install Debian VM from iso

**Expected behavior**
Se the installation setup

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: Debian 11/RaspiOS ETC
 - ARCH: RPI4B

Tacoos33, 2024-02-12:
On a raspberry pi 5 8Gb i have the same issue
When trying to start a Debian 12.5.0 arm64 iso: "guest has not initialized display yet"
When trying to start an Ubuntu 23.10 live server arm64 iso: "display output is not active"

Both machines are set-up as PiMox 8.3.1:
4 Gb mem, 1 socket 2 cores, Seabios, default (Virt) machine, VirtIO scisi single.
Doesn't seem to matter what type of display i choose, nor does increasing mem to a max of 512

nopmop, 2024-02-13:
You can get around this bug by launching the VM manually with a script (use SPICE as a display option).
I'm thinking to write a PR to fix this. 

Please post the output of: `qm showcmd <VMID> --pretty`
Thanks

muko82, 2024-02-13:
I was missing some step in the VM creation that I read from here: https://github.com/jiangcuo/Proxmox-Arm64/wiki/Qemu-VM
It seems that now everything is working almost fine.

I will need to boot from SSD instead of microSD: for test this is enough to me. :-)

closed 2024-02-13 by muko82

nopmop, 2024-02-13:
@muko82 did you manage to solve Display output with SeaBios?

muko82, 2024-02-13:
> @muko82 did you manage to solve Display output with SeaBios?

Unfortunately not... Just with uefi.

Tchoupinax, 2024-02-27:
Hello @muko82,
When you talk about booting from SSD instead of microSD, are you talking about your host or your VM?
Then, could you please provide a screenshot of your working configuration. For sure it could help me and next people who are searching about this issue.
Thank you a lot!