jiangcuo/pxvirt #55: 龙芯3c5000l+7a1000启动失败

yz03369343, 2024-03-01:
龙芯3c5000l+7a1000启动失败，咨询龙芯怀疑可能是内核bug导致，测试了龙芯的archlinux也有类似问题，但龙芯已经有了patch解决了改问题，还请评估是否能够合入到proxmox-ve的iso中

**硬件环境**
CPU：3c5000
主板型号：LS2C5LE
桥片：7A1000
固件版本：UDK2018-LoongArch-LS2C5LE-pre-beta9
[image]

**软件环境**
ISO镜像：proxmox-ve_8.1-3-loong64.iso
MD5:036efccdd36d53552ca33ac653846819

**故障现象**
grub界面后卡在Loading initial ramdisk
[image]

串口调试输出如下：
[image]

**龙芯patch链接**
https://github.com/loongarchlinux/linux/commit/3cb11d189673ae012ecad07bbbcd70bb08ca884b

jiangcuo, 2024-03-01:
Could you help me do a test ?

update linux to archlinux: 6.7.6  https://github.com/jiangcuo/pve-port-kernel/tree/loong64

The Kernel has  been upload . Download link:
`https://mirrors.apqa.cn/proxmox/debian/kernel/dists/sid/test/binary-loong64/`
1. You can unpack the deb file  and create a fat partition on usbdisk. 
2. Copy kernel image into usbdisk .
3. Poweron your sever and then enter efishell environment. 
4. Use "Direct Linux Boot" to start kernel. eg `vmlinux-6.7.6-pve console=ttyS0`
5. The kernel will be halt if this kernel is work. And I will create a new release .

yz03369343, 2024-03-04:
@jiangcuo 您好，感谢您的答复与指导

测试了您链接中的新内核，在安装iso中用新内核已经可以正常引导了，引导后卡在busybox这里，看起来内核是能正常运行了，附件是串口日志，请参考，再次感谢～
[Serial_log.txt](https://github.com/jiangcuo/Proxmox-Port/files/14475780/Serial_log.txt)

[image]

jiangcuo, 2024-03-04:
I will complete the update as soon as possible, it may take 2 weeks, because I am on a business trip now.

jiangcuo, 2024-03-16:
New iso upload
https://mirrors.lierfang.com/proxmox/isos/proxmox-ve_8.1-4-loong64.iso

yz03369343, 2024-04-23:
thinks～

closed 2024-04-23 by yz03369343