jiangcuo/pxvirt #58: RSP5 OSD Down after reboot

wuast94, 2024-03-13:
**Describe the bug**
When i Create an Ceph OSD it works without problems, as soon as i reboot the node the OSD wont come back up.
**To Reproduce**
Steps to reproduce the behavior:
1. Install Proxmox and Ceph (using your repos of course)
2. Create OSD
3. Reboot
4. OSD gone

**ENV (please complete the following information):**
 - OS: Debian GNU/Linux 12 (bookworm)
 - ARCH: [arm64
 - Raspberry PI 5
 ```
pveversion -v
proxmox-ve: 8.1.0 (running kernel: 6.6.20+rpt-rpi-v8)
pve-manager: 8.1.3+pve1 (running version: 8.1.3+pve1/26764642342c55bb)
proxmox-kernel-helper: 8.1.0
ceph: 18.2.0-pve2
ceph-fuse: 18.2.0-pve2
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx7
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.0.7
libpve-apiclient-perl: 3.3.1
libpve-common-perl: 8.1.0-1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.4
libpve-network-perl: 0.9.5
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.0.5
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.3
pve-cluster: 8.0.5
pve-container: 5.0.8
pve-docs: 8.1.3
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.1.4
pve-qemu-kvm: 8.1.2-4
pve-xtermjs: 4.16.0-3
qemu-server: 8.0.10+pve1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.0-pve4
```

**Additional context**

systemctl status ceph-osd@*.service dont give back anything
also journalctl -xeu ceph-osd@2.service no entries

i double checked that i am on the right host and use the right OSD number

jiangcuo, 2024-03-13:
/var/log/ceph is there has osd logs ?

wuast94, 2024-03-13:
more context: 

`lvs --version`
```
  LVM version:     2.03.16(2) (2022-05-18)
  Library version: 1.02.185 (2022-05-18)
  Driver version:  4.48.0
  Configuration:   ./configure --build=aarch64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/aarch64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --libdir=/lib/aarch64-linux-gnu --sbindir=/sbin --with-usrlibdir=/usr/lib/aarch64-linux-gnu --with-optimisation=-O2 --with-cache=internal --with-device-uid=0 --with-device-gid=6 --with-device-mode=0660 --with-default-pid-dir=/run --with-default-run-dir=/run/lvm --with-default-locking-dir=/run/lock/lvm --with-thin=internal --with-thin-check=/usr/sbin/thin_check --with-thin-dump=/usr/sbin/thin_dump --with-thin-repair=/usr/sbin/thin_repair --with-udev-prefix=/ --enable-applib --enable-blkid_wiping --enable-cmdlib --enable-dmeventd --enable-editline --enable-lvmlockd-dlm --enable-lvmlockd-sanlock --enable-lvmpolld --enable-notify-dbus --enable-pkgconfig --enable-udev_rules --enable-udev_sync --disable-readline
```

`lsblk`
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0   1.8T  0 disk 
nvme0n1     259:0    0 238.5G  0 disk
```

`vgchange -ay`

```
1 logical volume(s) in volume group "ceph-b9cc563f-5758-4ead-bbec-74c6aafb7099" now active
```
`lsblk` after `vgchange -ay`

```
NAME                                                                                                  MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda                                                                                                     8:0    0   1.8T  0 disk 
└─ceph--b9cc563f--5758--4ead--bbec--74c6aafb7099-osd--block--8331d767--af24--40da--bac0--ccbaf0fcda92 254:0    0   1.8T  0 lvm
```

`/var/lib/ceph/osd/ceph-2` is empty

wuast94, 2024-03-15:
Found a workaround:
After restart executing 
`vgchange -ay` activates the Logical Volumes and then all the automations take over

IF the restart was longer ago and the automations run into problems running `ceph-volume lvm activate --all`  afterwards brings the OSD back up again

adding
`@reboot /usr/sbin/vgchange -ay >> /var/log/vgchange.log 2>&1`
to my crontab fixes the issue for me

this is a workaround that fixes my specific error, but i think there is something off that also impacts hot plugging etc.

i hope all my information helps to get this fixed 😊

zafar-2020, 2025-01-12:
@jiangcuo I have exact same issue. While two osd are working fine. They are ssd. The last osd which is hdd works fine until I restart the system. In that case, the lvm is lost and it is in "Down" state. The above solution didn't work for me.

Any ideas? @wuast94

wuast94, 2025-01-13:
ive not using this anymore on my rsp`s, sry

closed 2025-04-23 by jiangcuo