jiangcuo/pxvirt #61: apt upgrade fails on debian 12.5 /  PVE 8.1.3+pve1

sipiyou, 2024-03-13:
Everything is running since some weeks fine as I used this howto to install everything on Pi4.

https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm

But today I tried to update the base debian system and I am getting this error:

Do I need to remove something ?

-----
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pve-firmware : Conflicts: firmware-linux-free but 20200122-1 is to be installed
                Conflicts: firmware-linux-free:armhf
E: Broken packages
---

This is what dpkg shows me:

root@pimox:/# dpkg -l | grep -i firmware
ii  bluez-firmware                       1.2-9+rpt2                          all          Firmware for Bluetooth devices
ii  pve-edk2-firmware-aarch64            4.2023.08-2                         all          edk2 based ARM64 UEFI firmware modules for virtual machines
ii  pve-edk2-firmware-legacy             4.2023.08-2                         all          edk2 based legacy 2MB UEFI firmware modules for virtual machines
ii  pve-edk2-firmware-ovmf               4.2023.08-2                         all          edk2 based UEFI firmware modules for virtual machines
ii  pve-firmware                         3.8-1                               all          Binary firmware code for the pve-kernel
ii  raspi-firmware                       1:1.20231024+ds-1+rpt5              all          Raspberry Pi family GPU firmware and bootloaders
ii  rpi-update                           20230904                            all          Raspberry Pi firmware updating tool

Thanks so far for your help :)

jiangcuo, 2024-03-15:
https://github.com/jiangcuo/Proxmox-Arm64/issues/19

sipiyou, 2024-03-15:
Thanks for your feedback but the problem is still there:

root@pimox:~# apt download pve-firmware
Get:1 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 pve-firmware all 3.8-1 [97.3 MB]
Fetched 97.3 MB in 34s (2,824 kB/s)                                                                                                                                                                                                                                                      
W: Download is performed unsandboxed as root as file '/root/pve-firmware_3.8-1_all.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
root@pimox:~# dpkg -i --force-overwrite pve-firmware_*.deb
(Reading database ... 94346 files and directories currently installed.)
Preparing to unpack pve-firmware_3.8-1_all.deb ...
Unpacking pve-firmware (3.8-1) over (3.8-1) ...
Setting up pve-firmware (3.8-1) ...
root@pimox:~# apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease                                                              
Get:2 http://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]                                           
Get:3 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]                          
Get:4 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]                                                
Get:6 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [144 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [374 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [383 kB]
Hit:5 https://mirrors.lierfang.com/proxmox/debian/pve bookworm InRelease                                                                                                                                                                                                                 
Fetched 1,027 kB in 14s (72.9 kB/s)                                                                                                                                                                                                                                                      
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
43 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@pimox:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pve-firmware : Conflicts: firmware-linux-free but 20200122-1 is to be installed
                Conflicts: firmware-linux-free:armhf
E: Broken packages
root@pimox:~#

jiangcuo, 2024-04-13:
It's been too long, close the issue

closed 2024-04-13 by jiangcuo