jiangcuo/pxvirt #63: debian sid install pve port(bookworm) error

inspireMeNow, 2024-03-15:
**Describe the bug**
I use pve bookworm ports to install because when I search the repo in (https://mirrors.apqa.cn/proxmox/), in sid it has only riscv and longarch builds. However, it has dependency errors.

**To Reproduce**
```bash
echo "deb https://mirrors.apqa.cn/proxmox/debian/pve bookworm port">/etc/apt/sources.list.d/pveport.list
echo "deb https://mirrors.apqa.cn/proxmox/debian/pve bookworm port">/etc/apt/sources.list.d/pveport.list
export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
apt update && apt install -y proxmox-ve
```

**Expected behavior**
Install successfully.

**Screenshots**
[image]

**ENV (please complete the following information):**
 - OS: debian sid
 - ARCH: arm64
 - Version: pve8
 - Machine: orangepi 5 plus 

**Additional context**
None.

jiangcuo, 2024-03-15:
Only riscv and loong64 should use sid.
If you're arm, it should be based on Debian bookworm and not sid

inspireMeNow, 2024-03-17:
Is there any way to install pve arm64 on debian sid? I can't downgrade to debian bookworm because it will broke many dependencies.

jiangcuo, 2024-03-19:
You can build pve from source.Only bookworm and bullseye support now.

closed 2024-03-20 by inspireMeNow