jiangcuo/pxvirt #64: GPG signing key curl command download empty file

UbioZur, 2024-03-15:
**Describe the bug**
Following the wiki or the repo.md file inside the help folder. The curl command
`curl https://mirrors.apqa.cn/proxmox/debian/pveport.gpg` download an empty file.

`curl -IL https://mirrors.apqa.cn/proxmox/debian/pveport.gpg` download the proper file.

**To Reproduce**
Steps to reproduce the behavior:
1. `curl https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o pveport.gpg`
2. cat pveport.gpg
3. File is empty

**Expected behavior**
I am no expert but following other answer from curl issues, it seems to be due to curl not following redirection, so -IL need to be added to the curl command.
So the wiki and the help/repo.md file should be fixed.

teranex, 2024-03-15:
I also just ran into this issue. Adding `-L` to curl fixed it indeed.

jiangcuo, 2024-03-16:
This issue should be that I have enabled the Cloudflare redirect policy yesterday. I've updated the wiki, thanks for the feedback

UbioZur, 2024-03-16:
This page also need to be changed (in case people use it instead of the wiki)

https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md

jiangcuo, 2024-03-17:
> This page also need to be changed (in case people use it instead of the wiki)
> 
> https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md

done

UbioZur, 2024-03-17:
Thanks, Closing the issue also.

closed 2024-03-17 by UbioZur