jiangcuo/pxvirt #65: RPI5 Wiki/doc fix needed /boot/xxx.txt -> /boot/firmware/xxx.txt

UbioZur, 2024-03-17:
**Describe the bug**
This is probably a change from rpi5 at some point. All the documentation ask to change /boot/xxx.txt when the file to change is in /boot/firmware/xxx.txt .

There is a warning message on the older path, so not a big deal, but it could be nice to update the documentation / wiki to state the correct file (For Raspberry pi 5)

**To Reproduce**
Steps to reproduce the behavior:
1. Raspberry Pi 5
2. New install of bookworms (I don't know about older ones)
3. `cat /boot/config.txt`

> DO NOT EDIT THIS FILE
> 
> The file you are looking for has moved to /boot/firmware/config.txt

4. `cat /boot/cmdline.txt`

> DO NOT EDIT THIS FILE
> 
> The file you are looking for has moved to /boot/firmware/cmdline.txt

**Expected behavior**
Up to date information on the wiki/docs.

**Affected Docs**
- https://github.com/jiangcuo/Proxmox-Port/wiki/RaspberryPi#2-ct-notes

  > nano /boot/cmdline.txt

to

  > nano /boot/firemware/cmdline.txt

- https://github.com/jiangcuo/Proxmox-Port/wiki/RaspberryPi#1-kernel-pagesize

The command/information should probably be written on the wiki (not behind a link)

> you should use the Kernel with 4K pagesize.
> 
> https://github.com/jiangcuo/Proxmox-Arm64/issues/65

to

> The Default Kernel for Pi5 use 16K and not all Software is compatible with this.
> As a Example proxmox-backup-client, he cant download previous Manifest files and must upload all chunks everytime.
> 
> Please modify config.txt like `nano /boot/config.txt`
> 
> Put the following parameters to the end of the file
> 
> `kernel=kernel8.img`
> 
> Ressources:
> https://www.raspberrypi.com/documentation/computers/config_txt.html
> https://github.com/jiangcuo/Proxmox-Arm64/issues/65

PS: I couldn't find a way to edit the wiki and offer a PR so I am making them as issue for the project itself. Thanks for the great work also.

gmacario, 2024-03-17:
I believe the change was done in bookworm: https://forums.raspberrypi.com/viewtopic.php?t=360134

jiangcuo, 2024-03-19:
I don't have a raspberry pi, I can't verify what you proposed. But I will update the wiki so more people can try it.

jiangcuo, 2024-03-20:
I re-researched this thread, the firmware issue is an old one. pve-firmware is used on x86, and it conflicts with other firmware.I don't have any good ideas at the moment, maybe I need to buy a Raspberry Pi.

Anexgohan, 2024-03-30:
can confirm atleast on the RPi5-8gb which i have the new location is 
/boot/firemware/

jiangcuo, 2024-03-31:
Update

https://github.com/jiangcuo/Proxmox-Port/wiki/RaspberryPi

UbioZur, 2024-04-02:
Seems like a better doc to me.

closed 2024-04-02 by UbioZur