jiangcuo/pxvirt #66: Not sure why this is happening.

mcooperRPI, 2024-03-17:
Hey Guys,

I am trying to create a VM on the host and I am getting this in the syslogs:

`Mar 17 05:17:39 opi5-pve-01 pvestatd[1949]: QEMU/KVM cannot detect CPU flags on ARM (aarch64)
Mar 17 05:17:39 opi5-pve-01 pvestatd[1949]: CPU flag detection failed, will try again after delay`

Can anyone help me please? I do not know how to set the CPU Flags to resolve this. 

This is what I am try to complete this on:
`Orange Pi 1.1.8 Bookworm with Linux 5.10.160-rockchip-rk3588`
 
And before you ask yes I installed everything via your instructions.
`root@opi5-pve-01:~# history
    1  nano -c /etc/network/interfaces
    2  cp /etc/network/interfaces /etc/network/interfaces.bak
    3  ls -al /etc/network/interfaces
    4  nano -c /etc/network/interfaces
    5  echo 'deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port'>/etc/apt/sources.list.d/pveport.list
    6  curl -L https://mirrors.apqa.cn/proxmox/debian/pveport.gpg -o /etc/apt/trusted.gpg.d/pveport.gpg 
    7  apt update && apt -y full-upgrade
    8  apt -y install ifupdown2
    9  apt -y install proxmox-ve postfix open-iscsi
   10  apt download pve-edk2-firmware-aarch64=3.20220526-rockchip
   11  dpkg -i pve-edk2-firmware-aarch64_3.20220526-rockchip_all.deb
   12  ls -al
   13  ping 192.168.2.8
   14  ping 192.168.3.9
   15  ip a
   16  nano -c /etc/network/interfaces
   17  sudo reboot
   18  ping 192.168.3.1
   19  ping 192.168.3.90
   20  ping 192.168.3.9
   21  ping 192.168.2.8
   22  history
   23  apt download pve-edk2-firmware-aarch64=3.20220526-rockchip
   24  dpkg -i pve-edk2-firmware-aarch64_3.20220526-rockchip_all.deb
   25  dpkg -i pve-edk2-firmware-aarch64_3.20220526-rockchip_all-1.deb`

Thanks guys,
Michael

jiangcuo, 2024-03-19:
ignore message

jiangcuo, 2024-04-13:
No update, close it

closed 2024-04-13 by jiangcuo