jiangcuo/pxvirt #70: Install failed (depends...)

hexakosi, 2024-03-23:
I followed the Manual on https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm

Raspberry pi, Debian bookworm minimal, fresh installed

The Command "apt install proxmox-ve postfix open-iscsi" broke with:
`root@home:/home/hexakosi# apt install proxmox-ve postfix open-iscsi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 proxmox-ve : Depends: pve-manager (>= 8.0.4) but it is not installable
              Depends: pve-qemu-kvm but it is not installable
              Depends: qemu-server but it is not installable
E: Unable to correct problems, you have held broken packages.`

Any Idea, what went wrong?

jiangcuo, 2024-03-27:
do
` apt policy`
`apt policy qemu-server`

totalizator, 2024-03-28:
Same here:
root@tinkerboard:~# apt policy qemu-server
qemu-server:
  Installed: (none)
  Candidate: (none)
  Version table:
root@tinkerboard:~# hostnamectl
The following packages have unmet dependencies:
 proxmox-ve : Depends: pve-manager (>= 8.0.4) but it is not installable
              Depends: pve-qemu-kvm but it is not installable
              Depends: qemu-server but it is not installable
E: Unable to correct problems, you have held broken packages.
root@tinkerboard:~# apt policy qemu-server
qemu-server:
  Installed: (none)
  Candidate: (none)
  Version table:
root@tinkerboard:~# hostnamectl
 Static hostname: tinkerboard
       Icon name: computer
      Machine ID: 1e161af9730c401283b9be11e62c4a61
         Boot ID: 1da2efc1c26b44588e37fa8f1b398488
Operating System: Armbian 24.2.5 bookworm
          Kernel: Linux 6.6.22-current-rockchip
    Architecture: arm

hexakosi, 2024-03-28:
thanks jiangcuo for your reply - does that say something to you?
is there an source package missing?

```
root@home:/home/hexakosi#  apt policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=arm64
     origin archive.raspberrypi.com
 500 http://archive.raspberrypi.com/debian bookworm/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.com
 500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     release n=bookworm,c=port,b=arm64
     origin mirrors.apqa.cn
 500 http://raspbian.raspberrypi.com/raspbian bookworm/rpi armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=rpi,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/non-free armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=non-free,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/contrib armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=contrib,b=armhf
     origin raspbian.raspberrypi.com
 500 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf Packages
     release o=Raspbian,a=stable,n=bookworm,l=Raspbian,c=main,b=armhf
     origin raspbian.raspberrypi.com
```

```
root@home:/home/hexakosi# apt policy qemu-server
qemu-server:arm64:
  Installed: (none)
  Candidate: 8.1.1+port2
  Version table:
     8.1.1+port2 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.1.1+port1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.10+pve1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.10 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.8-1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.7 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.6-1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.6 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
     8.0.6~1 500
        500 https://mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 Packages
```

```
root@home:/home/hexakosi# apt install qemu-server
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qemu-server:arm64 : Depends: genisoimage:arm64 but it is not installable
                     Depends: libjson-xs-perl:arm64 but it is not installable
                     Depends: libnet-ssleay-perl:arm64 but it is not installable
                     Depends: libpve-access-control:arm64 (>= 8.0.0~) but it is not installable
                     Depends: libpve-apiclient-perl:arm64 but it is not installable
                     Depends: libpve-cluster-perl:arm64 but it is not installable
                     Depends: libpve-common-perl:arm64 (>= 8.0.2) but it is not installable
                     Depends: libpve-guest-common-perl:arm64 (>= 5.0.3) but it is not installable
                     Depends: libpve-storage-perl:arm64 (>= 7.2-10) but it is not installable
                     Depends: libterm-readline-gnu-perl:arm64 but it is not installable
                     Depends: libuuid-perl:arm64 but it is not installable
                     Depends: libxml-libxml-perl:arm64 but it is not installable
                     Depends: perl:arm64 (>= 5.10.0-19) but it is not installable
                     Depends: proxmox-websocket-tunnel:arm64 but it is not going to be installed
                     Depends: pve-cluster:arm64 but it is not going to be installed
                     Depends: pve-firewall:arm64 but it is not going to be installed
                     Depends: pve-ha-manager:arm64 (>= 3.0-9) but it is not going to be installed
                     Depends: pve-qemu-kvm:arm64 (>= 7.1~) but it is not going to be installed
                     Depends: socat:arm64 but it is not installable
                     Depends: swtpm:arm64 but it is not going to be installed
                     Depends: swtpm-tools:arm64 but it is not going to be installed
                     Depends: libc6:arm64 (>= 2.34) but it is not going to be installed
                     Depends: libglib2.0-0:arm64 (>= 2.12.0) but it is not installable
                     Depends: libjson-c5:arm64 (>= 0.15) but it is not installable
                     Recommends: proxmox-backup-file-restore:arm64 (>= 2.1.9-2) but it is not going to be installed
                     Recommends: libpve-network-perl:arm64 (>= 0.8.3) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

my current sources:

>root@home:/home/hexakosi# cat /etc/apt/sources.list
>deb [ arch=armhf ] http://raspbian.raspberrypi.com/raspbian/ bookworm main contrib non-free rpi

>root@home:/home/hexakosi# cat /etc/apt/sources.list.d/
>pveport.list  raspi.list

>root@home:/home/hexakosi# cat /etc/apt/sources.list.d/pveport.list
>deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port

>root@home:/home/hexakosi# cat /etc/apt/sources.list.d/raspi.list
>deb http://archive.raspberrypi.com/debian/ bookworm main

hexakosi, 2024-03-29:
sorry, that was really stupid of me...
the fact that i've now made the mistake three times and haven't noticed that i'm using the 32bit image emphasizes the stupidity even more.

thank you for your patience and thank you for your work.
installation went smoothly with a 64bit image :)

greetings
hexakosi

closed 2024-03-29 by hexakosi