jiangcuo/pxvirt #71: Problems after recent update

napper306, 2024-03-26:
Good day.
First off thanks for the port. I've been using it for over a year with no particular issue, until the latest update.
Running 2 installs on Rapsberry Pi 4B, 8G models. I'm not sure where on Github to post the following, so I've decided to stick it in the Issues area.

**ENV**
 -OS: Debian GNU/Linux 12 (bookworm)
 -ARCH: RaspberryPi4 - arm64
 -Version 
proxmox-ve: 8.1.0 (running kernel: 6.6.20+rpt-rpi-v8)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.6
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: 4.2023.08-2
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.3-pve1

**Describe the bug**
With latest update, both installs have issues. First problem VMs would not launch, also could not perform backups. Error message was:

TASK ERROR: EFI base image '/usr/share/pve-edk2-firmware//AAVMF_CODE.ms.fd' not found

I resolved this by checking the read me at https://github.com/jiangcuo/pve-edk2-firmware/blob/master/debian/README.Proxmox-VE
I see under "AAVMF_CODE.ms.fd - This is a symlink to AAVMF_CODE.fd."
I used the following to make a symlink to the AAVMF_CODE.fd file. This fixed the error message and allows the expected behavior (VMs to boot again and able perform backups):

ln -s /usr/share/pve-edk2-firmware/AAVMF_CODE.fd /usr/share/pve-edk2-firmware/AAVMF_CODE.ms.fd

I'm not sure if this is the correct way to solve the problem, but it worked for me. If it is wrong, could you please advise what would be preferable.

**Describe the bug**
There is also another error message appearing in the Proxmox system log after the updates. It is the following:

Mar 26 07:37:04 Pimox pvestatd[1260]: VM -1 qmp command failed - VM -1 qmp command 'query-cpu-model-expansion' failed - The CPU type 'host' requires KVM
Mar 26 07:37:04 Pimox pvestatd[1260]: warning: failed querying supported tcg flags: VM -1 qmp command 'query-cpu-model-expansion' failed - The CPU type 'host' requires KVM
Mar 26 07:37:04 Pimox pvestatd[1260]: warning: failed querying supported kvm flags: QEMU flag querying VM exited with code 1 at /usr/share/perl5/PVE/QemuServer.pm line 3397.
Mar 26 07:37:04 Pimox pvestatd[1260]: CPU flag detection failed, will try again after delay

The "CPU flag detection failed" has been there since day 1, but the other warnings have not. Nothing seems to be failing because of this, but I don't know why the message is appearing and if it is a problem (aside from cluttering the system logs).

If there is remediation needed or any further information is required, let me know.
Thanks again for the project. It has been very useful to me.

jiangcuo, 2024-03-27:
All done. New packages uploaded

napper306, 2024-03-27:
I appreciate the quick response. The patches were applied. The error messages in the logs have now stopped. VMs and backups are behaving as expected. Very nice.
Have a great day.

closed 2024-03-30 by jiangcuo