jiangcuo/pxvirt #72: --scsi0 local:cloudinit won't work qm set or create

mzhaqs, 2024-03-27:
**Describe the bug**
--scsi0 local:cloudinit used to work fine before the recent update

**To Reproduce**
qm create 90107 --name pimox-1-kube-vm-1 --ostype l26 --cpu host --scsihw virtio-scsi-pci --scsi0 local:cloudinit --boot order=virtio0 --onboot 1 --bios ovmf --efidisk0 local:0

** Error **
unable to parse directory volume name 'cloudinit'

**Expected behavior**
VM should get created

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: Raspberry pi bookwork
 - ARCH: arm64
 - Version 8.1.7

**Additional context**
NA

jiangcuo, 2024-03-30:
This  issue exist on x86_64 too.

nikos445, 2024-04-01:
any workaround?

mzhaqs, 2024-04-02:
I just moved to 7 for now. But I believe rolling back to 8.1.3 will work as well.

nikos445, 2024-04-06:
I used Sata for cloud-init it worked (maybe a workaround for some people)...

Also when booting from ISO/CD then the SCSI of IMG that you boot shall be set to no. **2** when created or else it wont boot.

jiangcuo, 2024-04-13:
This issue has been fixed in the new commit.
https://git.proxmox.com/?p=qemu-server.git;a=commit;h=7e2065956f4cc5c15f947f0df431f46861b0a0c0

closed 2024-04-13 by jiangcuo

mzhaqs, 2024-04-16:
Issue seems to still exists
[image]

mzhaqs, 2024-04-16:
Issue seems to still exist. This is from a fresh installation.
[image]

SuperKali, 2024-04-26:
@mzhaqs @jiangcuo I have the same problem, on nanopc t6 only on the fresh installation the second node not have this problem

SuperKali, 2024-04-26:
[image]

SuperKali, 2024-04-26:
@jiangcuo @mzhaqs I had to compile myself the qemu-server and add this https://github.com/jiangcuo/qemu-server/commit/7e2065956f4cc5c15f947f0df431f46861b0a0c0 on the branch arm64, now work fine, can you apply this patch and release it on the repo?
[image]

For arm64 users can use this package compiled by me at the moment:
[qemu-server_8.1.1+port3_arm64.zip](https://github.com/jiangcuo/Proxmox-Port/files/15133404/qemu-server_8.1.1%2Bport3_arm64.zip)