jiangcuo/pxvirt #76: Proxmox Backup cron loop

SuperKali, 2024-04-10:
**Describe the bug**
Today I encountered a bug, I don't know why it happens, but I created a backup job every day at 9:00 p.m., I don't know what happens, but at the end of the backup after a few minutes it starts the job again as if it is in a loop

**To Reproduce**
Steps to reproduce the behavior:
1. I create a backup job
2. i configured a directory for host those backup
3. at 9:00 pm start automatically the backup job
4. after the backup end, 1/3 minutes it starts again.

**Expected behavior**
The correct working of backup system

**Screenshots**
[image]

[image]

[image]

[image]

**ENV (please complete the following information):**
 - OS: Debian 12
 - ARCH: ARM64
proxmox-ve: 8.1.0 (running kernel: 6.8.4-edge-rockchip-rk3588)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.3
pve-firmware: not correctly installed
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port2
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0

jiangcuo, 2024-04-11:
1. Disable sendmail
2. Install Postfix  and configure

SuperKali, 2024-04-11:
Hello @jiangcuo can i disable both and use notification through pve panel? And how this problem it's correlated

SuperKali, 2024-04-12:
Solved by disabled postfix

closed 2024-04-12 by SuperKali