jiangcuo/pxvirt #81: 虚拟机经过 clone 之后，无法启动，会停留在 EFI Shell

qiongzhu, 2024-04-16:
**Describe the bug**
近期更新较多包之后遇到此问题，之前没有。使用ISO安装的KVM虚机可以顺利启动。关闭虚机，然后从PVE网页界面里执行Clone，新虚机无法启动，会停留在 EFI Shell 上。

**To Reproduce**
Steps to reproduce the behavior:
1. 使用任意支持的 iso 安装一个 kvm 虚机，设置OMVF格式，创建EFI Disk，安装系统后，检查可以启动
2. 将该虚机正常关闭
3. 从PVE网页界面左侧的树状列表中选中该虚机，鼠标右键点击后选择 'Clone'，对话框中保持默认，点击'Clone'，等待操作完成
4. 选择新虚机，点击 Start后发现无法启动

存储使用 thin-lvm，相关软件版本是：
```
lvm2/stable,now 2.03.16-2 arm64 [installed]
thin-provisioning-tools/stable,now 0.9.0-2 arm64 [installed]
```

edk2 固件版本
```
pve-edk2-firmware-aarch64/unknown,now 4.2023.08-4 all [installed,automatic]
pve-edk2-firmware-legacy/unknown,now 4.2023.08-4 all [installed,automatic]
pve-edk2-firmware-ovmf/unknown,now 4.2023.08-4 all [installed,automatic]
```

**Expected behavior**
Clone后的虚机可以像原虚机一样启动

**Screenshots**
[image]

**ENV (please complete the following information):**
[image]

 - OS: debian12
 - ARCH: arm64
 - Version 
```
proxmox-ve: 8.1.0 (running kernel: 6.1.43-vendor-rk35xx)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.1.1+port2
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
```

**Additional context**
Add any other context about the problem here.

jiangcuo, 2024-04-16:
https://github.com/jiangcuo/Proxmox-Port/wiki/Rockchip-device-NOTE

qiongzhu, 2024-04-17:
前面我漏了升级前的版本信息。我之前用的 pve 版本大概是 8.1.3，这个版本非常稳定，保持了很长时间没有更新，使用上也没有问题。近期debian12和pve都有较多更新，升到了8.1.7后，发现拷贝后的虚机无法启动的问题。这个升级和FAQ里的7升8情况不同

安装 pve-edk2-firmware-aarch64 最新版本 4.2023.08-4时，从8.1.3时候留下的虚机，以及随后用ISO新安装的虚机，都可以启动。和FAQ中描述的似乎有些不同

现在我将 pve-edk2-firmware-aarch64 降级到指定版本，拷贝之后虚机无法启动的问题仍然存在，通过iso直接安装的都可以正常启动，其行为与 最新版本 4.2023.08-4 一致。
```
pve-edk2-firmware-aarch64/unknown,now 3.20220526-rockchip all [installed,upgradable to: 4.2023.08-4]
pve-edk2-firmware-legacy/unknown,now 4.2023.08-4 all [installed,automatic]
pve-edk2-firmware-ovmf/unknown,now 4.2023.08-4 all [installed,automatic]
```

jiangcuo, 2024-04-17:
那你降级edk2版本了吗？

jiangcuo, 2024-04-17:
那你降级试试，

closed 2024-04-24 by qiongzhu