jiangcuo/pxvirt #82: VM can't get an IP address from DHCP using a Linux Bridge Network

sebaxtian, 2024-04-17:
**Describe the bug**

I did just a fresh Proxmox installation on my Raspberry Pi 4 following the instructions below:
https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bullseye

I have a Raspberry Pi 4 with an 8 GB RAM

I going to describe step by step my configuration to reproduce the issue:

Step 1:
Create a Linux Bridge Network vmbr0 to my physical eth0 network:

[image]

Step 2:
Create a VM to install Debian 12 arm64 on it, I used the hardware configuration below:

[image]

Step 3:
During the Debian installation process on the VM I couldn't get network access, but I continued the process and finished the installation

Step 4:
I started the VM with Debian 12 arm64, then I did log in and I reconfigured the network interface to get a DHCP IP address from the Debian 12 arm64, I have the configuration below:

[image]

As you can see the issue related is getting a DHCP IP address from the VM with a Debian 12 arm64, this issue also happened with the Ubuntu Server, I don't know what is the problem related to the issue and I would like to know how to fix this issue, because I would like to get internet access from any Proxmox VM.

**Expected behavior**

The expected behaviour is to get a DHCP IP address from any Proxmox VM thus I can get internet access from any Proxmox VM

**ENV (please complete the following information):**

 - OS: debian11, where I already have installed Proxmox, I mean in my Raspberry Pi 4
  PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
  NAME="Debian GNU/Linux"
  VERSION_ID="11"
  VERSION="11 (bullseye)"
  VERSION_CODENAME=bullseye
  ID=debian
  HOME_URL="https://www.debian.org/"
  SUPPORT_URL="https://www.debian.org/support"
  BUG_REPORT_URL="https://bugs.debian.org/"
  
 - ARCH: arm64, where I already have installed Proxmox, I mean in my Raspberry Pi 4
 
 - Version: do `pveversion -v`, where I already have installed Proxmox, I mean in my Raspberry Pi 4
 
  proxmox-ve: 7.3-1 (running kernel: 6.1.21-v8+)
  pve-manager: 7.4-16 (running version: 7.4-16/0f39f621)
  pve-kernel-helper: 7.3-8
  ceph-fuse: 17.2.6-pve1
  corosync: 3.1.7-pve1
  criu: 3.16-2
  glusterfs-client: 9.2-1
  ifupdown: residual config
  ifupdown2: 3.2.0-1+pmx3
  libjs-extjs: 7.0.0-1
  libknet1: 1.24-pve2
  libproxmox-acme-perl: 1.4.4
  libproxmox-backup-qemu0: 1.3.2-1
  libproxmox-rs-perl: 0.2.1
  libpve-access-control: 7.4.1
  libpve-apiclient-perl: 3.2-1
  libpve-common-perl: 7.4-2
  libpve-guest-common-perl: 4.2-4
  libpve-http-server-perl: 4.2-3
  libpve-rs-perl: 0.7.7
  libpve-storage-perl: 7.4-3
  libqb0: not correctly installed
  libspice-server1: 0.14.3-2.1
  lvm2: 2.03.11-2.1
  lxc-pve: 5.0.2-2
  lxcfs: 5.0.3-pve1
  novnc-pve: 1.4.0-1
  proxmox-backup-client: 2.4.1-1
  proxmox-backup-file-restore: 2.4.1-1
  proxmox-mail-forward: 0.1.0-1
  proxmox-mini-journalreader: 1.3-1
  proxmox-widget-toolkit: 3.7.3
  pve-cluster: 7.3-3
  pve-container: 4.4-6
  pve-docs: 7.4-3
  pve-edk2-firmware: 3.20230228-4~bpo11+3
  pve-firewall: 4.3-4
  pve-firmware: not correctly installed
  pve-ha-manager: 3.6.1
  pve-i18n: 3.0.5
  pve-qemu-kvm: 7.2.0-8
  pve-xtermjs: 4.16.0-1
  qemu-server: 7.4-4+pve1
  smartmontools: 7.2-pve3
  spiceterm: 3.2-2
  swtpm: 0.8.0~bpo11+3
  vncterm: 1.7-1
  zfsutils-linux: 2.1.11-pve1

**Additional context**

This issue is really annoys me because I watch a lot of videos about how to create VMs and everything works well but when I try to create a VM in my Raspberry Pi 4 this scenario doesn't work.

This issue also happened with the Ubuntu Server, I don't know what is the problem related to the issue and I would like to know how to fix this issue, because I would like to get internet access from any Proxmox VM.

I really appreciate it if you could help me to solve this issue.

jiangcuo, 2024-04-19:
If docker is installed, uninstall it. If not,do `sysctl -a|grep bridge`.

grigio, 2024-04-24:
```
root@bbbb:~# sysctl -a|grep bridge
net.bridge.bridge-nf-call-arptables = 1
net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.bridge.bridge-nf-filter-pppoe-tagged = 0
net.bridge.bridge-nf-filter-vlan-tagged = 0
net.bridge.bridge-nf-pass-vlan-input-dev = 0

oot@bbbb:~# cat /etc/network/interfaces 
auto lo
iface lo inet loopback
iface lo inet6 loopback

auto eth0
iface eth0 inet dhcp

root@bbbb:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0@if47: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether bc:24:11:cb:cd:4e brd ff:ff:ff:ff:ff:ff link-netnsid 0

```

jiangcuo, 2024-04-26:
```
systemctl -w net.bridge.bridge-nf-call-arptables=0
systemctl -w net.bridge.bridge-nf-call-ip6tables=0
systemctl -w net.bridge.bridge-nf-call-iptables=0
```

grigio, 2024-04-26:
@jiangcuo Thanks, I think you mean: 

```
sysctl -w net.bridge.bridge-nf-call-arptables=0
sysctl -w net.bridge.bridge-nf-call-ip6tables=0
sysctl -w net.bridge.bridge-nf-call-iptables=0
```

I tried that but it seems I still can't get an DHCP ip from the outside router

sebaxtian, 2024-04-26:
> If docker is installed, uninstall it. If not,do `sysctl -a|grep bridge`.

Hi there, thank you so much @jiangcuo my issue was related as you mentioned, I had Docker installed on the host, Raspberry Pi 4 with Proxmox installed at the same time.

My solution was to install all the OS again and also install Proxmox again, this time without using Docker on the host, then everything worked well, I could create CTs with DHCP IP addresses.

I will go to share some screenshots later to document that DHCP IP addresses are assigned successfully.

Thank you

grigio, 2024-04-26:
@sebaxtian can you share your `/etc/network/interfaces` from the host and from the LXC container ?

sebaxtian, 2024-04-28:
> @sebaxtian can you share your `/etc/network/interfaces` from the host and from the LXC container ?

Hi @grigio `/etc/network/interfaces` from the host:

```shell
auto lo
iface lo inet loopback

iface eth0 inet manual

auto vmbr0
iface vmbr0 inet static
        address 192.168.1.99/24
        gateway 192.168.1.1
        bridge-ports eth0
        bridge-stp off
        bridge-fd 0

source /etc/network/interfaces.d/*
``` 

I'm using Ubuntu Jammy 22.04.4 LTS LXC template form here:
https://images.linuxcontainers.org/images/

Network interfaces from LXC Container:
```shell
root@ubuntu-jammy:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0@if4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether bc:24:11:c0:73:40 brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet 192.168.1.21/24 metric 1024 brd 192.168.1.255 scope global dynamic eth0
       valid_lft 86318sec preferred_lft 86318sec
    inet6 fe80::be24:11ff:fec0:7340/64 scope link 
       valid_lft forever preferred_lft forever
```

sebaxtian, 2024-04-28:
> Hi there, thank you so much @jiangcuo my issue was related as you mentioned, I had Docker installed on the host, Raspberry Pi 4 with Proxmox installed at the same time.
>
> My solution was to install [...]

Hello everyone, I want to share my new configuration. I already tested the new configuration on Raspberry Pi 3/4/5.

I created 1 LXC Container for each Raspberry Pi 3/4/5 using the Ubuntu Jammy 22.04.4 LTS LXC template from here:
https://images.linuxcontainers.org/images/

I have 3 hosts running Proxmox:

- Raspberry Pi 3
- Raspberry Pi 4
- Raspberry Pi 5

---

**Raspberry Pi 3**:
```bash
cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 3 Model B Rev 1.2
```
```bash
uname -m
aarch64
```
```bash
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
[image]

---

**Raspberry Pi 4**:
```bash
cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 4 Model B Rev 1.5
```
```bash
uname -m
aarch64
```
```bash
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
[image]

---

**Raspberry Pi 5**:
```bash
cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 5 Model B Rev 1.0
```
```bash
uname -m
aarch64
```
```bash
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
[image]

---

**Proxmox Network on host Raspberry Pi 3/4/5**:
[image]

---

**LXC Container on Raspberry Pi 3/4/5**:
[image]
[image]
[image]
[image]

---

**DHCP IP (192.168.1.21) on LXC Container on Raspberry Pi 3**:
[image]

---

**DHCP IP (192.168.1.19) on LXC Container on Raspberry Pi 4**:
[image]

---

**DHCP IP (192.168.1.20) on LXC Container on Raspberry Pi 5**:
[image]

---

**Ping Test for each LXC Container**:
[image]
[image]
[image]

---

Finally this issue is already solved and tested on Raspberry Pi 3/4/5.

---

**Tips**:
- Do not install Docker on the host
- Create a Linux Bridge Network
- When you create a LXC Container use default DHCP IP

---

Thank you @jiangcuo 

**CLOSED**

closed 2024-04-28 by sebaxtian

grigio, 2024-04-28:
> > @sebaxtian can you share your `/etc/network/interfaces` from the host and from the LXC container ?

Thanks a lot !!! I replicated your config and I'm able to run Proxmox arm64 like x86-64 !

The config was almost identical to mine, using Ubuntu22 instead of Debian I get the ip from the outside router!