jiangcuo/pxvirt #84: arm架构虚拟机在控制台操作时会异常关闭

yz03369343, 2024-04-23:
您好，我在arm架构上运行虚拟机时，当通过控制台持续操作约一段时间后虚拟机就会异常关闭，大约是1-5分钟左右，如果虚拟机启动后不放在控制台上，采用ssh访问过去，则不会异常关闭；系统中也未发现有明显的异常日志，初步怀疑还是和控制台的某些机制有关，如同某个操作把当前控制的虚拟机进程kill掉了一样，请问大佬，这个问题您是否有遇到？是否有解决方案呢？

使用的ISO：proxmox-ve_8.1-3_arm64.iso
CPU：HUAWEI Kunpeng 920 5221K
服务器信息：百信 Hengshan TS02F-F30

yz03369343, 2024-04-23:
pve组件信息如下

root@pve-kp-01:~# pveversion -v
proxmox-ve: 8.1.0 (running kernel: 6.1.62-generic)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.62-generic: 6.1.62
pve-kernel-5.15.138-generic: 5.15.138
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx7
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.25-1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.0-1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.4
libpve-network-perl: 0.9.5
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.0.5
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
openvswitch-switch: 3.1.0-2
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.3
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.0.10+pve1
smartmontools: 7.3-1+b1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.0-pve4

jiangcuo, 2024-04-26:
https://github.com/jiangcuo/Proxmox-Port/issues/15

closed 2024-05-20 by yz03369343