jiangcuo/pxvirt #89: 请求内核添加对 ebpf 的更多支持

bigoovip, 2024-05-09:
我目前在准备搭建 dae 服务，dae 需要以下内核选项：
```
CONFIG_BPF=y
CONFIG_BPF_SYSCALL=y
CONFIG_BPF_JIT=y
CONFIG_CGROUPS=y
CONFIG_KPROBES=y
CONFIG_NET_INGRESS=y
CONFIG_NET_EGRESS=y
CONFIG_NET_SCH_INGRESS=m
CONFIG_NET_CLS_BPF=m
CONFIG_NET_CLS_ACT=y
CONFIG_BPF_STREAM_PARSER=y
CONFIG_DEBUG_INFO=y
# CONFIG_DEBUG_INFO_REDUCED is not set
CONFIG_DEBUG_INFO_BTF=y
CONFIG_KPROBE_EVENTS=y
CONFIG_BPF_EVENTS=y
```

而本项目中内核不支持以下选项
```
CONFIG_BPF_STREAM_PARSER=y
CONFIG_DEBUG_INFO=y
# CONFIG_DEBUG_INFO_REDUCED is not set
CONFIG_DEBUG_INFO_BTF=y
```

能否考虑增加以上选项到内核？

jiangcuo, 2024-09-19:
本项目没有为rockchip维护内核。你应该提交到rockchip官方。

jiangcuo, 2025-01-17:
在默认的内核中，加入了BPF的支持，因为rockchip并不是本人维护，关闭改case
https://github.com/jiangcuo/pve-port-kernel/commit/59aa42423dcba728c037a0f45be75021099d7d32

closed 2025-01-17 by jiangcuo