jiangcuo/pxvirt #90: Raspberry Pi 5 - NVME Pimoroni adapter not detected

sitymarz, 2024-05-10:
Trying to use a Raspberry Pi 5 with the Pimoroni NVME adapter.

Doesn't look like Proxmox detects the device.
When testing with ESXI installer, this does detect the NVME drive to install to

 - OS: proxmox-ve_8.1-7_arm64
 - ARCH: arm64 Raspberry Pi 5

Anexgohan, 2024-08-12:
make sure you have enabled the pcie interface in

Add to bottom of 
`/boot/firmware/config.txt`

```
# Enable pcie
dtparam=pciex1

# Note: You could also just add the following (it is an alias to the above line)
# dtparam=nvme

# Optionally, you can control the PCIe lane speed using this parameter, 3 sets to pcie gen 3 speed 10G/s, 2 is 5G/s
dtparam=pciex1_gen=3
```

closed 2025-04-23 by jiangcuo

hfagelnour, 2025-05-20:
> make sure you have enabled the pcie interface in
>
> Add to bottom of `/boot/firmware/config.txt`
>
> ```
> # Enable pcie
> dtparam=pciex1
>
> # Note: You could also just add the following (it is an alias to the [...]

I used these settings and no matter what i do the installer does not detect the NVME ssd installed on ARGON ONE V5 NVME adapter! the weird thing, the NVME SSD is detected in the boot devices inside the UEFI devices list!
help please