jiangcuo/pxvirt #96: Ubuntu Noble != 24.04?

Miq1, 2024-06-05:
**Describe the bug**
When attempting to create a container on an Raspberry Pi 5 with the ubuntu noble template the process is stopped with an error:
```
()
Formatting '/var/lib/vz/images/102/vm-102-disk-0.raw', fmt=raw size=34359738368 preallocation=off
Creating filesystem with 8388608 4k blocks and 2097152 inodes
Filesystem UUID: f4f04eaf-83a2-4a54-a9ef-141b756d8870
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624
extracting archive '/var/lib/vz/template/cache/ubuntu-noble-rootfs.tar.xz'
Total bytes read: 570132480 (544MiB, 62MiB/s)
TASK ERROR: unable to create CT 102 - unsupported Ubuntu version '24.04'
```

asylumexp, 2024-06-06:
proxmox v8.2 is required (pve-container update fixes this)

jiangcuo, 2024-06-13:
> proxmox v8.2 is required (pve-container update fixes this)

done

asylumexp, 2024-06-14:
confirmed working now

closed 2024-06-30 by jiangcuo