jiangcuo/pxvirt #99: Partition does not exist - Raspberry pi 5 Imager

DanieleMassa, 2024-06-16:
After downloading the ISO from this repo and flashing it with the official raspberry tool, I get "Partition does not exist" (in the image below). I've already tried to format the sdcard using gparted before flashing it, nothing changed tho. For reference, I'm using a rpi5 with 8GB of ram (powered by the official power supply) and a SDSQXA1-128G-GN6MN as sdcard
[image]

SuperKali, 2024-06-17:
@DanieleMassa There are no specific images for proxmox at the moment, use armbian and install proxmox via repository.
- https://github.com/jiangcuo/Proxmox-Port/wiki/Install-or-upgrade-from-repo
- https://github.com/armbian/community/releases/download/24.8.0-trunk.139/Armbian_community_24.8.0-trunk.139_Rpi5b_bookworm_current_6.6.33_minimal.img.xz

DatHardie, 2024-09-19:
> @DanieleMassa There are no specific images for proxmox at the moment, use armbian and install proxmox via repository.
>
> * https://github.com/jiangcuo/Proxmox-Port/wiki/Install-or-upgrade-from-repo
> * [...]

Is something similar as the armbian installer existing for the Raspberry Pi 4?

DanieleMassa, 2024-09-19:
> Is something similar as the armbian installer existing for the Raspberry Pi 4?

the armbian installer works also for rpi4, check [this link](https://www.armbian.com/rpi4/). If you want to install proxmox on it, check [this guide](https://github.com/DanieleMassa/proxmoxOnRPI5) I've written, should work fine also on rpi4

closed 2024-09-19 by DanieleMassa