jiangcuo/pxvirt #100: Using Ceph pool for CT fails - rbd kernel module missing

snooops, 2024-06-21:
**Describe the bug**
After the installation of the proxmox 8 following the bookworm guide in this wiki, i followed those steps: https://pve.proxmox.com/wiki/Deploy_Hyper-Converged_Ceph_Cluster to setup a ceph cluster. Ceph cluster is up and running among three Raspberry PI 5 8GB systems.
If i move a disk from local storage into the ceph storage i get the error message: 
```
modinfo: ERROR: Module rbd not found.
modprobe: FATAL: Module rbd not found in directory /lib/modules/6.6.31+rpt-rpi-2712
rbd: failed to load rbd kernel module (1)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install Proxmox 8 using this guide: https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm on three raspberry pi 5 8gb using the sdcard and raspbian 8 bookworm 
2. Create a Proxmox Cluster using the webinterface for those three raspberries.
3. Follow this guide to create a proxmox cluster: https://pve.proxmox.com/wiki/Deploy_Hyper-Converged_Ceph_Cluster
4. On the first node: do the ceph init, skip that step on the other two nodes
5. create on each node a ceph mon
6. create on each node a osd device
7. create a pool using `pveceph pool create piceph --add_storages`
8. Create a CT and assign local storage
9. Once CT is created, try to move CT storage from local storage to piceph storage.

**Expected behavior**
ceph storage works without error.

**ENV (please complete the following information):**
 - OS: debian 12 - Linux loki 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
 - ARCH: arm64
 - Version 
```
proxmox-ve: 8.2.0 (running kernel: 6.6.31+rpt-rpi-2712)
pve-manager: 8.2.4 (running version: 8.2.4/0098b77b88c86db8)
proxmox-kernel-helper: 8.1.0
ceph: 17.2.6-pve1+3
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.7
libpve-cluster-perl: 8.0.7
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-network-perl: 0.9.8
libpve-rs-perl: 0.8.9
libpve-storage-perl: 8.2.1
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: not correctly installed
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.4-pve1
```

**Additional context**
Add any other context about the problem here.

jiangcuo, 2024-06-25:
Update your kernel to support rbd

snooops, 2024-06-25:
I googled around and saw this: https://github.com/raspberrypi/linux/issues/6224 
I guess in a few days a new kernel with rbd support will come automatically to raspberry pi.

closed 2024-06-30 by jiangcuo