jiangcuo/pxvirt #105: libasan8 breaks qemu

soxrok2212, 2024-07-01:
**Describe the bug**
I have Debian 12 + Proxmox running on a Turing Pi RK1 on kernel 6.10 The recent libasan8 dependency suddenly crashes QEMU after I did an `apt update && apt upgrade`. Prior to this, everything was working as expected. Now, trying to start a VM results in:

```
==493126==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.
command '/usr/bin/kvm --version' failed: exit code 1

TASK ERROR: Detected old QEMU binary ('unknown', at least 3.0 is required)
```

LXC appears unaffected.

**To Reproduce**
Update to the latest from repo. Restart a VM, or just run `kvm` via a shell.

**Expected behavior**
VM should start up normally.

**Screenshots**
N/A

**ENV (please complete the following information):**
 - OS: Debian 12
 - ARCH: arm64, rk3588
 - Version:
 ```
proxmox-ve: 8.2.0 (running kernel: 6.10.0-rc1+)
pve-manager: 8.2.4 (running version: 8.2.4/0098b77b88c86db8)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.7
libpve-cluster-perl: 8.0.7
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-network-perl: 0.9.8
libpve-rs-perl: 0.8.9
libpve-storage-perl: 8.2.1
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: 3.8-1
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.4-pve1
```

jiangcuo, 2024-07-02:
Fresh install proxmox-ve from ISO proxmox-ve_8.1-7_arm64.iso and then upgrade, no of the above happens

soxrok2212, 2024-07-02:
Odd issue that was solved with the following:

```
sudo touch /please-remove-proxmox-ve
sudo apt purge proxmox*
sudo apt autoremove
sudo apt install proxmox-ve
```

The uninstall scripts may complain about a missing Proxmox script, copy it from another system and paste it in then try to uninstall again.

closed 2024-07-02 by soxrok2212