jiangcuo/pxvirt #109: Orange Pi 5 | MAC address changes on reboot

dandycyclone, 2024-07-11:
**Describe the bug**
Full description could be found [here](https://github.com/MichaIng/DietPi/issues/6663), since I raise this issue is a duplicate to backport the fix into current project. In a nutshell:
 - eth mac address changes after each SBC reboot - it could be confirmed with the kernel logs:

```
root@pve:~# dmesg | grep rk_get_eth_addr
[   42.710816] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: rk_vendor_read eth mac address failed (-1)
[   42.710828] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: use serial to generate eth mac address: 00:00:a4:bd:7f:ff
[   42.710839] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: mac address: 00:00:a4:bd:7f:ff
```

**To Reproduce**
Steps to reproduce the behavior:
1. Boot PVE into `Orange Pi 5` with `6.1.43-rockchip` kernel.
2. Record eth mac address and reboot the board.
3. Verify, that eth mac address was changed after reboot.

**Expected behavior**
eth mac address should be persistent across reboots. Current (non-expected) behavior results in problems with DHCP IP reservation, when eth_mac is used as DHCP Client ID.

**Screenshots**
 - N/A

**ENV (please complete the following information):**
 - OS: _Debian GNU/Linux_
 - ARCH: _arm64_
 - Version: _12 (bookworm)_

**Additional context**
 - _proxmox-ve_ was booted using EDK2 UEFI firmware for Rockchip RK3588 platforms: https://github.com/edk2-porting/edk2-rk3588

dandycyclone, 2024-07-11:
It is also worth mentioning this information from https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#supported-peripherals-in-uefi, however not saying it should be the source of problem:

> GMAC Ethernet |  Not working | Only brought-up for OS usage

x13-me, 2024-11-07:
This is inherent to the RK3588S built-in ethernet, you can simply add `hwaddress <10:DC:B6:90:00:00 - 10:DC:B6:9F:FF:FF>` to the `/etc/network/interfaces` file as normal, or configure a bridge

dandycyclone, 2024-11-07:
What does this part of your comment mean, @x13-me?
> This is inherent to the RK3588S built-in ethernet

Do you want to say, reported behavior is "_expected_" for RK3588S even though this expected behavior was addressed in https://github.com/MichaIng/DietPi/issues/6663#issuecomment-2108580052 ?

dandycyclone, 2024-11-07:
Yes, agree. Thank you very much for explanation.

dandycyclone, 2024-11-07:
Works as intended/designed.
Moreover for Proxmox PVE it is not really a problem, that might be of a big operational concern.

closed 2024-11-07 by dandycyclone