jiangcuo/pxvirt #112: arm64版本虚拟机使用cloud-init后开机报错

zhufeng, 2024-07-15:
**Describe the bug**
鲲鹏920的宿主机，虚拟机在添加cloud-init设备之后开机报错。看了下像是-ide虚拟设备的问题。

**To Reproduce**
Steps to reproduce the behavior:
1. 创建一台虚拟机。
2. 添加一个cloud-init设备并开机。
3. 开机报错。
4. 报错日志如下：
```
generating cloud-init ISO
WARN: no efidisk configured! Using temporary efivars disk.
kvm: -device ide-cd,bus=ide.0,unit=0,drive=drive-ide0,id=ide0: Bus 'ide.0' not found
TASK ERROR: start failed: QEMU exited with code 1
```

**Expected behavior**
正常使用cloud-init设备。

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ENV (please complete the following information):**
 - OS: debian 12
```
[root@TG225-PVE ~]# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```
 - ARCH: arm64
 - Version [ do `pveversion -v`]
```
proxmox-ve: 8.2.0 (running kernel: 6.1.62-generic)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
pve-kernel-6.1.62-generic: 6.1.62
pve-kernel-5.15.138-generic: 5.15.138
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
ksmtuned: 4.20150326+b1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.1
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.4
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.2.1
libpve-guest-common-perl: 5.1.3
libpve-http-server-perl: 5.1.0
libpve-network-perl: 0.9.8
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 6.0.0-1
lxcfs: 6.0.0-pve2
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.2.5-1
proxmox-backup-file-restore: 3.2.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.2.3
pve-cluster: 8.0.7
pve-container: 5.1.11
pve-docs: 8.2.2
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.7
pve-firmware: 3.8-1
pve-ha-manager: 4.0.5
pve-i18n: 3.2.2
pve-qemu-kvm: 8.1.5-4
pve-xtermjs: 5.3.0-3
qemu-server: 8.2.1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.4-pve1

```

**Additional context**
尝试过使用scsi设备来加载cloud init的设备，但是arm64平台似乎只支持使用ide设备。

jiangcuo, 2024-07-29:
Update your proxmox-ve version.

bakeDong1, 2024-09-23:
cloud init 设备使用scsi 可以解决

closed 2025-04-23 by jiangcuo