jiangcuo/pxvirt #113: Bug on apt update

jhmc93, 2024-07-20:
I Get the following the error when I run apt update: `Err:6 https://mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease
  521   [IP: 104.21.43.231 443]
Reading package lists... Done
E: Failed to fetch https://mirrors.apqa.cn/proxmox/debian/pve/dists/bookworm/InRelease  521   [IP: 104.21.43.231 443]
E: The repository 'https://mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

Vallentinus, 2024-07-20:
I have this problem too, It says 521 (Web Server Down). Maybe they are maintaining server or something. I hope they will fix soon.

mzarka-kku, 2024-07-21:
same issue for me. But I found that they provide other mirror: (https://github.com/jiangcuo/Proxmox-Port/wiki/Proxmox%E2%80%90Port--Repo-List)

Golbal: https://global.mirrors.apqa.cn (Cloudflare)
Korea: https://mirrors.apqa.cn (Cloudflare)
Hong Kong: https://hk.mirrors.apqa.cn (direct)
China: https://mirrors.lierfang.com (direct)
Germany: https://de.mirrors.apqa.cn (direct)

I test the last one, and it was working.

jhmc93, 2024-07-21:
> same issue for me. But I found that they provide other mirrors: (https://github.com/jiangcuo/Proxmox-Port/wiki/Proxmox%E2%80%90Port--Repo-List)
> 
> Golbal: https://global.mirrors.apqa.cn (Cloudflare) Korea: https://mirrors.apqa.cn (Cloudflare) Hong Kong: https://hk.mirrors.apqa.cn (direct) China: https://mirrors.lierfang.com (direct) Germany: https://de.mirrors.apqa.cn (direct)
> 
> I have just tested the last one, and it was working.

How would I go about importing this

napper306, 2024-07-26:
Having a similar issue. It seems like the website’s SSL certificate is expired, which is the message I'm getting when trying to open any of the mirrors in my browser as well. Seems like jiangcuo hasn't been around github much for a couple months. Hopefully he's doing OK. I get a lot of use from this project in my homelab. Here's a screenshot of my errors.
[image]

jiangcuo, 2024-07-29:
I need to pay for the server, and if it's in arrears, it will cause the site to fail
If someone want sync and share it , pls open a issue and i will put it on mirrors list.

napper306, 2024-07-29:
Glad you're doing OK jiangcuo. There's no way for people to donate/support the project available? Maybe some users have a way to contribute financially. I believe I recall someone asking previously about using this project in a business setting and wanting to pay. Thanks again for the work.

jhmc93, 2024-07-29:
> I need to pay for the server, and if it's in arrears, it will cause the site to fail If someone want sync and share it , pls open a issue and i will put it on mirrors list.

I’m sorry if you’re struggling to keep it going, I hope there’s some way u could get some support on this. If I was knowledgeable I’d of co hosted it with you but I’m not great haha. Thanks for letting me know

xifi-kif, 2024-07-31:
> I need to pay for the server, and if it's in arrears, it will cause the site to fail If someone want sync and share it , pls open a issue and i will put it on mirrors list.

I could set a mirror with https in EU, please provide some documentation/how-to to set the mirror up

napper306, 2024-08-08:
Changed my repository from the global mirror- https://global.mirrors.apqa.cn to https://mirrors.apqa.cn and the system is no longer complaining about an invalid certificate. Thanks again. I've got so many services in my home lab using Proxmox running on my Pi4 (PBS port, 6 containers, several VMs). It feels essential to my projects and I cried a little when the updates showed errors.

napper306, 2024-08-13:
> > Changed my repository from the global mirror- https://global.mirrors.apqa.cn to https://mirrors.apqa.cn and the system is no longer complaining about an invalid certificate. Thanks again. I've got so many services in my home lab using Proxmox running on my Pi4 (PBS port, 6 containers, several VMs). It feels essential to my projects and I cried a little when the updates showed errors.
> 
> how do I change the repo?

One way you could change it is using a text editor like nano - for instance using the following command - nano /etc/apt/sources.list.d/pveport.list - change the repository location in that file then save it. Then you can refresh packages and upgrade from the Proxmox GUI or using apt update && apt upgrade. You initially added that repo when installing Proxmox Port with the command "echo 'deb [arch=arm64] https://mirrors.apqa.cn/proxmox/debian/pve bookworm port'>/etc/apt/sources.list.d/pveport.list". That echo command is another way to insert text into a file.
Though it appears from your initial comment above you already have the repository enabled that I'm having success with, so if you still are getting errors I'm not sure why.

closed 2025-04-23 by jiangcuo

topa-LE, 2025-06-06:
Ign:5 https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease
Fehl:5 https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease
Could not handshake: Ein fataler TLS-Alarm wurde empfangen.

Mirror again unavailable. SSL certificate invalid. Project is great, how can I support you?

7onnie, 2025-06-11:
Changing the Source to "mirrors.qpqa.cn" instead of "global.mirrors.apqa.cn" fixed it for me.

topa-LE, 2025-06-12:
No, it just doesn't work. I've tried everything. Maybe the developer can tell you how it currently works again.

`Holen:5 https://mirrors.apqa.cn/proxmox/debian/devel bookworm InRelease [2.621 B]
Holen:6 https://mirrors.apqa.cn/proxmox/debian/devel bookworm/port arm64 Packages [3.679 kB]
Es wurden 3.682 kB in 2 s geholt (1.607 kB/s).
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Aktualisierung für 1 Paket verfügbar. Führen Sie »apt list --upgradable« aus, um es anzuzeigen.
W: Das Laden der konfigurierten Datei »port/binary-armhf/Packages« wird übersprungen, da das Depot »https://mirrors.apqa.cn/proxmox/debian/devel bookworm InRelease« die Datei scheinbar nicht bereitstellt. (Schreibfehler bei der Angabe der Komponente in sources.list?)
root@pimox8:~#
`