jiangcuo/pxvirt #115: Random crashes

Toastyyy3, 2024-07-28:
**Describe the bug**
Every once in a while Proxmox crashes. I can't access any VM, LXC or the host itself afterwards until I cut power and turn it back on.

**To Reproduce**
No reproducible behavior observed. Times differ from mid night to mid day, journals don't show any errors. Maybe some of the containers corrupt the host? Raspberry Pi is located in a media shelf nearby the router, modem, Playstation 5 and Nintendo Switch. Or maybe something interferes with the Pi and leads to bit flips or something which then leads to a crash? 

VM: Home Assistant
LXC:
- Nextcloud
- Haproxy
- Adguard
- Zigbee2Mqtt
- MQTT Broker

Another o

**Expected behavior**
Not crash unpredictably.

**ENV (please complete the following information):**
 - OS: Raspi debian 12 (raspi.debian.net)
 - ARCH: arm64
 - Version 
<details><summary>Details</summary>
<p>

proxmox-ve: 8.1.0 (running kernel: 6.1.0-21-arm64)
pve-manager: 8.1.7 (running version: 8.1.7/ee1c3736ef6a6541)
proxmox-kernel-helper: 8.1.0
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx8
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.1
libpve-access-control: 8.1.3
libpve-apiclient-perl: 3.3.1
libpve-cluster-api-perl: 8.0.5
libpve-cluster-perl: 8.0.5
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.6
libpve-rs-perl: 0.8.7
libpve-storage-perl: 8.1.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.2-1
proxmox-backup-file-restore: 3.0.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.2
proxmox-widget-toolkit: 4.1.5
pve-cluster: 8.0.5
pve-container: 5.0.9
pve-docs: 8.1.4
pve-edk2-firmware: not correctly installed
pve-firewall: 5.0.3
pve-firmware: 3.8-1
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
[... 89 lines cut ...]
Jun 27 21:58:05 rpi4 pveproxy[1084]: starting 1 worker(s)
Jun 27 21:58:05 rpi4 pveproxy[1084]: worker 3916398 started
Jun 27 22:17:01 rpi4 CRON[3921398]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun 27 22:17:01 rpi4 CRON[3921399]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Jun 27 22:17:01 rpi4 CRON[3921398]: pam_unix(cron:session): session closed for user root
Jun 27 23:17:01 rpi4 CRON[3937313]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun 27 23:17:01 rpi4 CRON[3937314]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Jun 27 23:17:01 rpi4 CRON[3937313]: pam_unix(cron:session): session closed for user root
Jun 27 23:19:24 rpi4 pveproxy[3916398]: worker exit
Jun 27 23:19:24 rpi4 pveproxy[1084]: worker 3916398 finished
Jun 27 23:19:24 rpi4 pveproxy[1084]: starting 1 worker(s)
Jun 27 23:19:24 rpi4 pveproxy[1084]: worker 3937974 started
Jun 27 23:19:39 rpi4 pveproxy[3913339]: worker exit
Jun 27 23:19:39 rpi4 pveproxy[1084]: worker 3913339 finished
Jun 27 23:19:39 rpi4 pveproxy[1084]: starting 1 worker(s)
Jun 27 23:19:39 rpi4 pveproxy[1084]: worker 3938018 started
Jun 27 23:25:58 rpi4 pveproxy[3906891]: worker exit
Jun 27 23:25:58 rpi4 pveproxy[1084]: worker 3906891 finished
Jun 27 23:25:58 rpi4 pveproxy[1084]: starting 1 worker(s)
Jun 27 23:25:58 rpi4 pveproxy[1084]: worker 3939690 started

</p>
</details> 

Maybe someone can push me in the right direction...? Grateful for any kind of help!

jiangcuo, 2024-07-29:
Didn't see any useful information

napper306, 2024-07-29:
Have you tried plugging the Pi into a monitor after it becomes unavailable to see if it has actually crashed or if only network access is a problem?
Other possibilities: 
Resource allocation - too much RAM and/or CPU being assigned to VMs/containers, system seizing up from not having adequate resources available.
Drive issue, such as - USB can't keep up with I/O required or drive has bad sectors.
Power issues. Things connected via USB drawing more power than Pi can safely provide.

Toastyyy3, 2024-08-23:
Sorry, I wanted to reply sooner.

So I believe I fixed it and it wasn‘t a port issue. I installed the proxmox port on top of raspi debian and then I found somewhere (unfortunately can‘t find the post anymore) that something in the debian kernel had to be disabled to run proxmox stably I believe.

In other words, when installing Proxmox on top of Debian, something in the kernel or similar has to be disabled.

So issue closed, thanks for the help anyways!

closed 2024-08-23 by Toastyyy3